darkxst/silabs-firmware-builder #110: ZBDongle-E and ZB GreenPower devices

mti-sk, 2024-08-09:
Hi,
I use ZBDongle-E as a coordinator in ZB2MQQT in Ember mode (adapter: ember)
Therefore, it is necessary to use firmware 7.4.X (EZSP13) and higher.
Unfortunately, all 7.4.X firmwares seem to contain a bug (or lack of functionality) for ZB GreenPower devices.
The ZB GreenPower device (in my case the PowerTag A9MEM1570 Schneider) must be connected to the ZB2MQQT via the ZB GreenPower Proxy (in my case the Ikea Bulb TRADFRI LED2103G5). Unfortunately, ZBDongle-E cannot handle ZB GreenPower communication.
Pairing ends with the message: **error: zh:controller:greenpower: Error: Parameter 'tempMaster' is missing**
It's probably a firmware bug.
If I use ZBDongle-P (based on CC2652P) as a coordinator, I don't have this problem, pairing works well.

darkxst, 2024-08-09:
I have never tested any Green Power devices. It is quite possible that the firmware could be missing some required component for this to work. I will try take a look at this at some stage.

senna1992, 2024-08-13:
Same here, getting the "tempMaster" issue on ember driver, using HA yellow with its internal radio.
No difference if pairing is allowed via ikea light driver or "all"

mti-sk, 2024-08-30:
Hi,
unfortunately, I have to confirm that the problem: 
error: zh:controller:greenpower: Error: Parameter 'tempMaster' is missing 
persists even in the new firmware based on Gecko4.4.4: ncp-uart-hw-v7.4.4.0-zbdonglee-115200 .gbl
Maybe it will work next time...

darkxst, 2024-08-30:
I havent made any changes to enable Greenpower yet (assuming that is the issue). I will try get some devices to test though.

mti-sk, 2024-08-30:
Hi,
fingers crossed, but mostly thanks for the effort.

senna1992, 2024-08-30:
Can you get some if I support you with 50$\€? Or shall I buy some unavailable in your region and post them to you?

mti-sk, 2024-09-03:
it's there. I appreciated your effort.

mti-sk, 2024-09-03:
Hi,
This seems to fix: https://github.com/Koenkk/zigbee-herdsman/issues/1151- Ember: fix GP proxied messages handling. solved, at least partially solved ZGP device pairing (at least for now: ZB2MQTT v1.40.0, ZBDongle-E fw7.4.4 ).
My PowerTag A9MEM1570 can now be paired via GPP to the network with the ZBDongle-E coordinator (V2 model, EFR32MG21) .
Problems with the missing parameter :greenpower: Error: Parameter 'tempMaster' is missing still persist, but seem to be non-fatal now and the A9MEM1570 is communicating with the network. Thank you all for your efforts.

senna1992, 2024-09-03:
> Hi, This seems to fix: [Koenkk/zigbee-herdsman#1151](https://github.com/Koenkk/zigbee-herdsman/pull/1151) Ember: fix GP proxied messages handling. solved, at least partially solved ZGP device pairing (at least for now: ZB2MQTT v1.40.0, ZBDongle-E fw7.4.4 ). My PowerTag A9MEM1570 can now be paired via GPP to the network with the ZBDongle-E coordinator (V2 model, EFR32MG21) . Problems with the missing parameter :greenpower: Error: Parameter 'tempMaster' is missing still persist, but seem to be non-fatal now and the A9MEM1570 is communicating with the network. Thank you all for your efforts.

cool, looking forward to this being solved too! 
if testing is required i have some r9mem 1phase powertag to test the support.

did you know, that the new ikea smart plugs "tretakt" does have cluster 242 for "gp input" and "gp output" while tradfi versions had "gp output" only?

also the heiman hs2sk which are whitelabeld for schneider-wiser system, also offer "gp output" cluster only?
(this explains why all my powertags connect to hub v2 directly. (is readable when using wiser drayton integration from astanga)

mti-sk, 2024-09-03:
Hi,
Did I understand correctly that you are able to pair ZBGP device directly with the ZBDongle-E coordinator?
I have always needed a ZB GPP (Ikea power plug or Ikea light bulb)