darkxst/silabs-firmware-builder #111: Unable to flash firmware for Sonoff Dongle E

mckrosrw, 2024-08-11:
Hi there,

I am very new to this. I bought a dongle recently and read that it is advised to update the firmware. I tried https://darkxst.github.io/silabs-firmware-builder/ but it could not detect the firmware. I tried https://github.com/NabuCasa/universal-silabs-flasher, but I get a lot of error messages as per below. Any idea how I can resolve this? I am on a Windows laptop and a novice in this area:

[image]

If I can't get this to work, I guess I could try manually disassembling the dongle and using the reset bootloader (once I buy a screwdriver small enough for the screws :)). Otherwise, maybe the dongle just is not working.

thanks for any help

darkxst, 2024-08-11:
This error probably indicates an issue with the installation of the flasher. Did you install it using `pip`?

mckrosrw, 2024-08-12:
Thanks, yes I installed with pip. Should I remove the universal-silabs-flasher packlage and reinstall or could there be any other cause of the errors?

darkxst, 2024-08-12:
You could try reinstall it, however I not very familiar with windows myself. 

If you are using Home Assistant you could also try flashing using the Silabs Flasher Add-on

mckrosrw, 2024-08-13:
Yes, installed a more up-to-date version of python for windows (https://www.python.org/downloads/windows/), but after a  lot of time still getting the same error. Also, tried on a separate Windows machine without success. I did not know about the add-on, thanks for that. I will that a go before disassembling the sonoff dongle. Thanks again!

Nosnahc, 2024-10-20:
> Yes, installed a more up-to-date version of python for windows (https://www.python.org/downloads/windows/), but after a lot of time still getting the same error. Also, tried on a separate Windows [...]

Have you tried with [this windows driver](https://learn.adafruit.com/how-to-install-drivers-for-wch-usb-to-serial-chips-ch9102f-ch9102/windows-driver-installation)?

darkxst, 2024-10-20:
> Have you tried with [this windows driver](https://learn.adafruit.com/how-to-install-drivers-for-wch-usb-to-serial-chips-ch9102f-ch9102/windows-driver-installation)?

That is the driver for the original Dongle-E. Now they are using Silicon Labs cp2102 chips.

closed 2026-01-06 by github-actions[bot]