darkxst/silabs-firmware-builder #112: Unable to flash my Sonoff ZBdongle-E V1

ThibLeS-59, 2024-08-16:
I'm a beginner, and I can't seem to flash my ZBDongle-E V1. I first tried using the DARKXST site, where it was recognized, but the flashing failed. 

[silabs_flasher (2).log](https://github.com/user-attachments/files/16638559/silabs_flasher.2.log)

Then, I tried using the command line with universal_silabs_flasher, but the installation always ends in an error:
`universal-silabs-flasher --device COM5 --bootloader-reset sonoff flash --firmware "C:\.......`

[image]

tl-sl, 2024-08-17:
Try without bootloader reset, that seems to be causing the error

```
universal-silabs-flasher --device COM5  --firmware "C:\.......

```

ThibLeS-59, 2024-08-18:
Yes, i have a same error

EmmaPj18, 2024-08-27:
I have same issue 
[Log](https://github.com/user-attachments/files/16766257/silabs_flasher.log)

EmmaPj18, 2024-08-28:
@darkxst Now it can connect and detect the dongle E, but it fails trying to upload the new firmware

[image]

Logs: 
[silabs_flasher (1).log](https://github.com/user-attachments/files/16785097/silabs_flasher.1.log)

darkxst, 2024-08-29:
> I'm using Brave browser

Probably not, it seems the Windows usb serial driver is causing the dongle to reset when we open serial connection. I will see if I can workaround this second failure also.

AlphaArs, 2024-08-30:
I'm getting the same problem, just bought the Sonoff ZBDongle-E EFR32MG21.
I installed the  CP210x drivers and on the normal website I did get it detected. But when trying to update or change the firmware I get the error message. I tried both with the silabs builder and the cp-test.

here are the logs:
[patched flasher.log](https://github.com/user-attachments/files/16818469/patched.flasher.log)
[silabs_flasher.log](https://github.com/user-attachments/files/16818474/silabs_flasher.log)

EmmaPj18, 2024-08-30:
> Probably not, it seems the Windows usb serial driver is causing the dongle to reset when we open serial connection. I will see if I can workaround this second failure also.

@darkxst I tested with the original web patcher on a Macbook and it detects the dongle and I can flash the dongle successfully. As you said, this is a Windows issue with the driver

AlphaArs, 2024-08-30:
Thats interesting, so If I try to do this with a Linux vm it should work ? @EmmaPj18

EmmaPj18, 2024-08-30:
> Thats interesting, so If I try to do this with a Linux vm it should work ? @EmmaPj18

Idk but It could work

darkxst, 2024-08-30:
> so If I try to do this with a Linux vm it should work

yes I believe it will work on Linux. Also if you have a HA box, the Silabs Flasher Addon should work also.

madeira3003, 2024-09-14:
> > então se eu tentar fazer isso com uma VM Linux deve funcionar
> 
> sim, acredito que funcionará no Linux. Além disso, se você tiver uma caixa HA, o Silabs Flasher Addon também deve funcionar.

Can you help us on how to do this?

madeira3003, 2024-09-15:
> > so If I try to do this with a Linux vm it should work
> 
> yes I believe it will work on Linux. Also if you have a HA box, the Silabs Flasher Addon should work also.

Can you help us on how to do this?

Suffice, 2024-09-22:
I'm curious why I received an E V1 dongle when everyone else seems to have an E V2? It seems to be working fine with the latest Ember firmware, aside from my 1st gen Hue bulbs being almost always non-responsive.

I was able to flash my device on Linux, both in the browser and from the terminal.

darkxst, 2024-09-22:
V1 dongle is actually ZBDongle-P (cc2652). 
V2 dongle is ZBDongle-E.
Sonoff recently changed the USB chip in the ZBDongle-E, and at the time due to production issues, inadvertently flashed some dongles with the USB config file for ZBDongle-P, thus the USB product strings are wrong on these.

darkxst, 2024-11-09:
Continue in #107

closed 2024-11-09 by darkxst