darkxst/silabs-firmware-builder #116: Gecko 4.4.4

kalpik, 2024-08-21:
Hi!

Any plans for new builds? https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.4.4

Thanks!

darkxst, 2024-08-22:
Test builds are in this branch, I will merge to main branch after some testing.
https://github.com/darkxst/silabs-firmware-builder/tree/4.4.4

dspalu32, 2024-08-22:
zb-gw04-1v2/ncp-uart-hw-v7.4.4.0-zb-gw04-1v2-115200.gbl is working OK for me on two separate installations. Thanks!

johnsmoses, 2024-08-28:
Hello,

Just picked up a zbdongle-e and saw the recent build. Adding devices seems to work fine and i see no issues beyond the typical route error notices that the ember driver surfaces. Not sure if this is relevant but the dongle itself crashes and resets when trying to do an ota for a single device. Not sure if super critical as repeating the updates succeeds eventually. Attaching the log in any case. 

Used: ncp-uart-hw-v7.4.4.0-zbdonglee-115200.gbl

`[2024-08-28 18:17:54] debug: 	zhc:ota:common: Payload offsets: start=39840 end=39888 dataSize=48
[2024-08-28 18:17:54] debug: 	zh:controller:endpoint: CommandResponse 0x282c02bfffeb2705/1 genOta.imageBlockResponse({"status":0,"manufacturerCode":4659,"imageType":54182,"fileVersion":268513360,"fileOffset":39840,"dataSize":48,"data":{"type":"Buffer","data":[5,144,79,154,108,6,188,135,192,70,30,241,238,11,232,41,132,0,201,32,132,0,48,101,129,96,81,160,4,144,231,157,4,238,33,192,0,52,1,160,5,161,9,162,81,163,6,144]}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"writeUndiv":false})
[2024-08-28 18:17:54] debug: 	zh:ember: ~~~> [ZCL to=34686 apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":1,"direction":1,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":167,"commandIdentifier":5}]
[2024-08-28 18:17:54] debug: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=145 Len=86]
[2024-08-28 18:17:54] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=1 frmRx=1](ackRx=1)
[2024-08-28 18:17:54] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-08-28 18:17:54] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=1 frmTx=2)
[2024-08-28 18:17:54] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=1](frmRx=1) Added to rxQueue
[2024-08-28 18:17:54] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2](ackRx=1)
[2024-08-28 18:17:54] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=144 Len=44]
[2024-08-28 18:17:54] debug: 	zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":1,"options":320,"groupId":0,"sequence":185}], [packetInfo:{"senderShortId":34686,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":255,"lastHopRssi":-19,"lastHopTimestamp":0}], [messageContents=019503023312a6d35030011040980000300000]
[2024-08-28 18:17:54] debug: 	zh:controller: Received payload: clusterID=25, address=34686, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=255, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":149,"commandIdentifier":3},"payload":{"fieldControl":2,"manufacturerCode":4659,"imageType":54182,"fileVersion":268513360,"fileOffset":38976,"maximumDataSize":48},"command":{"ID":3,"response":5,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35},{"name":"fileOffset","type":35},{"name":"maximumDataSize","type":32}],"name":"imageBlockRequest"}}
[2024-08-28 18:17:54] debug: 	z2m: Received Zigbee message from 'Upper Hall Outlet', type 'commandImageBlockRequest', cluster 'genOta', data '{"fieldControl":2,"fileOffset":38976,"fileVersion":268513360,"imageType":54182,"manufacturerCode":4659,"maximumDataSize":48}' from endpoint 1 with groupID 0
[2024-08-28 18:17:55] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-08-28 18:17:55] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=1 frmTx=2)
[2024-08-28 18:17:55] debug: 	zh:ember:uart:ash: Timer expired waiting for ACK for frmTx=2, ackRx=1
[2024-08-28 18:17:55] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2](ackRx=1)
[2024-08-28 18:17:55] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA_RETX frmReTx=1 frmRx=2](ackRx=1 frmTx=2)
[2024-08-28 18:17:55] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-08-28 18:17:55] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=1 frmTx=2)
[2024-08-28 18:17:55] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2](ackRx=1)
[2024-08-28 18:17:56] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-08-28 18:17:56] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=1 frmTx=2)
[2024-08-28 18:17:56] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=2](frmRx=2) Added to rxQueue
[2024-08-28 18:17:56] debug: 	zh:ember:uart:ash: Timer expired waiting for ACK for frmTx=2, ackRx=1
[2024-08-28 18:17:56] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA_RETX frmReTx=1 frmRx=3](ackRx=1 frmTx=2)
[2024-08-28 18:17:56] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=144 Len=44]
[2024-08-28 18:17:56] debug: 	zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":1,"options":320,"groupId":0,"sequence":202}], [packetInfo:{"senderShortId":34686,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":255,"lastHopRssi":-19,"lastHopTimestamp":0}], [messageContents=01a603023312a6d3503 [... cut ...]
[2024-08-28 18:18:31] debug: 	z2m: Saving state to file /config/zigbee2mqtt/state.json
[2024-08-28 18:18:31] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2024-08-28 18:18:31] info: 	z2m: Disconnecting from MQTT server
[2024-08-28 18:18:31] info: 	z2m: Stopping zigbee-herdsman...
[2024-08-28 18:18:31] debug: 	zh:controller:database: Writing database to '/config/zigbee2mqtt/database.db'
[2024-08-28 18:18:31] info: 	z2m: Stopped zigbee-herdsman
[2024-08-28 18:18:31] info: 	z2m: Stopped Zigbee2MQTT `

closed 2026-01-05 by github-actions[bot]