darkxst/silabs-firmware-builder #123: Support for Silabs emberznet 7.4.4.0 and 8.0.0 versions

IgnacioHR, 2024-09-12:
It looks like there is a new version 7.4.4 (and there is also an 8.0.0) available.

Should they be added to firmware_build folder?

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.4.4.0.pdf
https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-8.0.0.0.pdf

Rgds

Xploder, 2024-09-14:
7.4.4 is available in the 4.4.4 branch and currently in testing
https://github.com/darkxst/silabs-firmware-builder/tree/4.4.4

8.0.0 had a serious showstopping bug causing instability that was supposedly fixed with 8.0.1

darkxst, 2024-09-14:
I will add 8.0.1 builds in the next week

Xploder, 2024-09-20:
New version 8.0.2 seems to have fixed LQI!

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-8.0.2.0.pdf

Xploder, 2024-09-29:
In the meantime we can grab a 8.0.2 based firmware here https://github.com/Nerivec/silabs-firmware-builder/releases

CodeForLove83, 2024-10-08:
@darkxst when do you plan to release firmware build 8.0.2 for zbdonglee? 
In addition, do you have any walkthrough to compile the firmware build for zbdonglee?

IgnacioHR, 2024-10-10:
I updated my coordinator to 8.0.2 yesterday and everything works the same as before. Means, AFAICS nothing is broken with 8.0.2

MSL-DA, 2024-10-10:
> I updated my coordinator to 8.0.2 yesterday and everything works the same as before. Means, AFAICS nothing is broken with 8.0.2

Same experience here

```
Zigbee2MQTT docker> 1.40.2-dev
Home Assistant > 2024.10.1
ZBDongle-E > 8.0.2
```

darkxst, 2024-10-20:
I have uploaded v8.0.2 build here for all supported adapters:
https://github.com/darkxst/silabs-firmware-builder/releases/tag/v20241020-beta

Nosnahc, 2024-10-20:
[image]
Don't work for me with zbdonglee_zigbee_ncp_8.0.2.0_sw_flow_115200.gbl 
I'm maybe doing this wrong.

darkxst, 2024-10-20:
Web flasher needs to be updated to be able to read the installed v8 firmware details (universal-silabs-flasher cli tool can read it though). You should just try test it Z2M etc

dspalu32, 2024-10-20:
v8.02 working well for me on two separate z2m setups. (zb-gw04-1v2_zigbee_ncp_8.0.2.0_115200.gbl)

The "Silicon Labs Flasher" HA add-on doesn't accept the v8 .gbl as a custom firmware file - where v7.x used to work - but a manual update using your webflasher works as expected. The python silabs flasher was OK for v8 also. The error from the HA add-on related to the firmware having v2 manifest format, where the add-on only supports v1. Presumably it does a cross-check of the target rather than flashing blindly, and will itself need an update.

Thanks!

darkxst, 2024-10-20:
> Presumably it does a cross-check of the target rather than flashing blindly, and will itself need an update.

Yes Silabs flasher relies on reading the firmware version first (so it doesnt reflash if the version is already installed.)  I will try get this updated.

darkxst, 2024-10-20:
Yes, Silabs flasher will also need an update to deal properly with v8 firmware

wastez, 2024-10-22:
You can use ember-zli to flash the firmware, i have done this for my dongle-e because the silabs flasher doesn’t work.

darkxst, 2024-11-02:
Update builds for 8.0.2 here:
https://github.com/darkxst/silabs-firmware-builder/releases/tag/20241102

lyonelf, 2024-11-03:
which file a must use for the dongle E in coordinator mode inside Jeedom ?

MSL-DA, 2024-11-03:
> which file i must use for the dongle E in coordinator mode inside Jeedom ?

https://github.com/darkxst/silabs-firmware-builder/releases/download/20241102/zbdonglee_zigbee_ncp_8.0.2.0_sw_flow_115200.gbl (Zigbee)

tootla, 2024-11-04:
> Updated builds for 8.0.2 here: https://github.com/darkxst/silabs-firmware-builder/releases/tag/20241102

@darkxst Based on your practical experience, what are the differences between Gecko SDK and Simplicity SDK?

darkxst, 2024-11-04:
Simplicity SDK is just the evolution of Gecko SDK, essentially just a name change but also dropping support for older chips (only Elelabs devices affected,  of the device I build for). However that resulted in lots of changes in itself,  so its hard to pick out actual practical improvements in v8. 

Otherwise usual applies if you want the latest and greatest install v8.0.2, if you just want something stable stick with v7.4.x series for now as that is relatively stable and mature at this point.

IgnacioHR, 2024-11-04:
Closing as completed! Thanks everyone

tore1980, 2024-11-04:
I have updated ZBdonle to 8.2.0 [zbdonglee_zigbee_ncp_8.0.2.0_sw_flow_115200.gbl] from 7.4.3, now I can`t see devises or cordinator in Z2M but it does work from HA. Has anyone experienced the same?
[image]

Z2M Config:
ort: >-
  /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220813141617-if00
adapter: ember
baudrate: 115200
rtscts: false

acommuni, 2025-01-16:
After firmware update (Sonoff Dongle E -> NCP 8.0.2.0) there were some errors with OTA requets but after a switch off/switch on, all seems to be good.

acommuni, 2025-01-16:
There is a new version 8.1 : https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-8.1.0.0.pdf

darkxst, 2025-01-17:
8.1.0 release is buggy, likely will wait for 8.1.1 before releasing this.

acommuni, 2025-02-09:
Here is the 8.1.1.0 : https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-8.1.1.0.pdf

darkxst, 2025-02-11:
Initial builds... I have had to disable software flow control on all devices that lack hw flow control.

https://github.com/darkxst/silabs-firmware-builder/releases/tag/20250211-Next

acommuni, 2025-02-23:
With 8.0.2, It works well for OTA update which takes more than 15mn whereas it does not works with firmware 7.4.5.0 or 7.4.4.0. But interview is not working for some devices (Sonoff SNZB-02D), it is better than previously because no interview was working. This time, reconfigure is not working for Sonoff SNZB-02D. (It is working working with 7.4.5)

The only log is the following for interview :

[2025-02-23 18:43:16] error: 	z2m: Request 'zigbee2mqtt/bridge/request/device/interview' failed with error: 'interview of 'XXXXXXXXXXXXXXX' (0xYYYYYYYYYYYYYYYYYY) failed: Error: Interview failed because can not get active endpoints ('0xYYYYYYYYYYYYYYY')'

For reconfigure :

[2025-02-23 18:48:19] error: 	z2m: Failed to configure 'XXXXXXXXXXXXXXX', attempt 1 (Error: Bind 0xYYYYYYYYYYYYYYY/1 genPowerCfg from '0x7cc6b6fffefb44df/1' failed (Delivery failed for '33960'.)
    at EmberOneWaitress.deliveryFailedFor (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:100:31)
    at EmberAdapter.onMessageSent (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:457:42)
    at Ezsp.emit (node:events:524:28)
    at Ezsp.ezspMessageSentHandler (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5055:14)
    at Ezsp.callbackDispatch (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:746:26)
    at Ezsp.onAshFrame (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:396:22)
    at UartAsh.emit (node:events:524:28)
    at Immediate.<anonymous> (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1171:45)
    at processImmediate (node:internal/timers:491:21))

How can I help ?

FPSUsername, 2025-05-07:
Seems like 8.0.3 as well as 8.1.0, 8.1.1 and 8.1.2 exist.

https://docs.silabs.com/d/zigbee-stack-api/8.1.2/version-history