darkxst/silabs-firmware-builder #124: cannot updqte the firmware of my SONOFF Zigbee E

imTHAI, 2024-09-13:
Hello,

I tried under windows 11 with edge (I aslo tried with chrome to be sure)

First I tried with the [regular website](https://darkxst.github.io/silabs-firmware-builder/):
I get this error:
[image]
And here are the debug logs:  https://gist.githubusercontent.com/imTHAI/2cf9a493a57dd2435ba060190e5a6426/raw/124aab25637fc1544b86a698730192e0868313db/gistfile1.txt

Then I tried with [an alternative website](https://darkxst.github.io/cp-test/) you mentioned in another opened issue:
With this website I go further:
[image]
But then when I click upgrade it stays at 0% for a few seconds then gives me the same error as the first website.
Here are the debug logs: https://gist.githubusercontent.com/imTHAI/bb2cfe637819308791979a33db551d13/raw/090cf70e12cf21d7c5143925d7caad7d34bf01a1/gistfile1.txt

Could you check and help me ?

Thank you

cornrow8, 2024-09-13:
same here ! 
pressing the boot button inside the key does not change anything !

cornrow8, 2024-09-13:
OK ! 
The easiest solution is to flash it within HA silicon labs flash addon.
It works like a charm!

imTHAI, 2024-09-13:
> OK ! 
> 
> The easiest solution is to flash it within HA silicon labs flash addon.
> 
> It works like a charm!
> 
> 

Can you explain to me or give me a link ? Silicon like in Apple silicon ?

thib3113, 2024-09-14:
Same issue here . I just buy a ZBDongle-E from amazon . 

HA addon silicon labs flash (no multi protocol), tell me he can't do nothing ... 

I found a russian video allowing to flash my firmware to 7.1.4 . 

---

When trying the alternative website, I can get the first popin, and it fail to update the firmware . 

In the devtools, I saw the error : 
```
pyodide.asm.js:9 Uncaught (in promise) PythonError: Traceback (most recent call last):
  File "/lib/python3.11/site-packages/universal_silabs_flasher/flasher.py", line 305, in enter_bootloader
    await self.probe_app_type(types=[ApplicationType.GECKO_BOOTLOADER])
  File "/lib/python3.11/site-packages/universal_silabs_flasher/flasher.py", line 261, in probe_app_type
    raise RuntimeError("Failed to probe running application type")
RuntimeError: Failed to probe running application type
```

wich seems to be the same error I got from flashing it manually with universal-silabs-flasher .

madeira3003, 2024-09-14:
> A solução mais fácil é fazer o flash dentro do HA s

Can you help us on how to do this?

imTHAI, 2024-09-14:
I also tried connecting to the key through console (SecureCRT). I tried holding the boot button, then reset button, then both etc.. I followed several tutorials. The light never blinks red ! The light stays red. For a very short time it switches to green (AND red, both colors in same time). 

Long story short, I think my key is defect. Back to Amazon.

thib3113, 2024-09-14:
@imTHAI it works on my key . But I didn't check the key leds ... (just, after upload, it doesn't show the menu like said, I just hit enter, and it's fine) . 

https://sonoff.tech/wp-content/uploads/2023/02/SONOFF-Zigbee-3.0-USB-dongle-plus-firmware-flashing.pdf

madeira3003, 2024-09-15:
> OK ! The easiest solution is to flash it within HA silicon labs flash addon. It works like a charm!

Can you help us on how to do this?

darkxst, 2024-09-15:
Install "Silabs Flasher Addon" from HA Addons page. Once installed there is a switch to show "unused configuration", this enables a custom url option that you can put github (raw) links into for flashing

madeira3003, 2024-09-15:
hardware flow control do I leave it active?

Verbose mode, no?

madeira3003, 2024-09-15:
[image]

darkxst, 2024-09-15:
hardware flow control should be off for ZDongle-E

madeira3003, 2024-09-15:
How do I know if the firmware is up to date?

darkxst, 2024-09-16:
you should be able to check firmware version in ZHA or Z2M.

Alternatively just try flash with silabs flasher, it will show versions in the log, and should not actually re-flash if you trying to install same version.

imTHAI, 2024-09-18:
Eventually I updated the key with the plugin "HA Silabs firmware upgrade" !

closed 2024-09-18 by imTHAI