darkxst/silabs-firmware-builder #125: Differences from stock Sonoff firmware & changelog

maxromanovsky, 2024-09-15:
@darkxst thank you very much for you work, especially for the web based firmware update tool.
I am new to the Zigbee world, and to the Silabs-based donlges as well.
I am using `ZBDongle-E`, hence all I wrote is related to it.
But I think it would be really nice to clarify:
- What's the difference between the firmware that you're building here and stock firmware provided by the dongle manufacturer? As of time of writing, it's [`7.4.3` for ZBDongle-E](https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master/Dongle-E/NCP_7.4.3)
- Why does your web update tool recommends a bit older version (`7.3.1.0`), even if newer firmwares are also available from your github repo?
- Where can I find the changelog for each build? Is it just the [gecko_sdk](https://github.com/SiliconLabs/gecko_sdk/releases) changelog, or do you have some secret sauce? :)

Thank you in advance for clarifications.
I think it makes sense to add them to the `README`.

darkxst, 2024-09-16:
I dont know what settings Itead use in their firmware builds, but probably similar to mine. For a very long time Sonoff didnt offer any recent builds at all. I do enable token interface which is used by ZHA backups (but not z2m just yet)

Webflasher has been updated to 7.4.4 today.

Gecko changelogs cover most updates for bug fix releases. There are upcoming build system changes coming though, that likely change a few settings.

maxromanovsky, 2024-09-16:
Thank you for the clarifications!

closed 2024-09-16 by maxromanovsky