darkxst/silabs-firmware-builder #130: Cant flash Zigbee Dongle-E

lennremy, 2024-09-23:
Hi,

Im currently still running my Dongle-E on the depricated ezsp adapter and wanted to update the firmware to use the ember adapter. But the flasher always shows the MacOS driver bug warning, although I've installed the driver to my mac.

I did attached the log, anyone can help me solve this issue?

[silabs_flasher.log](https://github.com/user-attachments/files/17098549/silabs_flasher.log)

hassio-med74, 2024-10-12:
Hello lennremy. Did you manage to solve the problem? Because I have the exact same issue. Thank you in advance.

MSL-DA, 2024-10-12:
I have the same problem. Mac Mini M1 with Sequoia

lennremy, 2024-10-12:
@hassio-med74 @MSL-DA kinda fixed it. I was able to update the dongle to a newer firmware, but I had to use a different method. Found this video that worked for me. Hope this helps

https://www.youtube.com/watch?v=BU9ha0Uaduc&t=161s

MSL-DA, 2024-10-12:
Thanks for the answer 👍 Can I get you to reopen this issue? I don't know if it is a driver issue. Maybe darkxst has something to add to that.

darkxst, 2024-10-12:
I have not previously heard of issues on Mac OS, it shouldnt be affected like the driver issues on windows.

The logs on OP are overly brief, the process seems to have ended prematurely, and nothing useful has been logged.

Can you test using the python `universal-silabs-flasher`, this might provide more useful logs.

https://pypi.org/project/universal-silabs-flasher/

MSL-DA, 2024-10-12:
I can ... I will receive an extra Sonoff Dongle-E sometime next week. I will return with a log file.

MSL-DA, 2024-10-13:
I can add that updating from EZPZ to ember 8.0.2 through SecureCRT went without error on macOS 15.0.1

darkxst, 2024-10-13:
well that is not python based, so doesnt help much in working out why webflasher not work. universal-silabs-flasher is what is used by the web flasher behind the scenes.

darkxst, 2024-10-13:
Also worth checking if there are any errors in the developer console logs (that didnt make it into downloadable logs)

FrancoisLef, 2024-10-19:
Hello 👋 

I am facing the same issue on Mac OS Sequoia 15.0.1.

- ✨ Brand new Sonoff Dongle E
- ✅ Drivers installed
- ♻️ Rebooted multiple times

I can help on testing / QA if needed 👍 

## Debug file

[silabs_flasher.log](https://github.com/user-attachments/files/17446811/silabs_flasher.log)

## MacOS driver

<p align="center">
  [image]
</p>

## Selected device

<p align="center">
  [image]
</p>

## Developer console logs

<p align="center">
  [image]
</p>

lroehrs, 2024-10-23:
As workaround I used:

https://pypi.org/project/universal-silabs-flasher/

```
$ universal-silabs-flasher --device /dev/cu.SLAB_USBtoUART \
  --ezsp-baudrate 115200 \
  flash \
  --firmware ../tmp1/silabs-firmware-builder/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.4.0-zbdonglee-115200.gbl
```
I got the firmware from here:
https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds

`ncp-uart-hw-v7.4.4.0-zbdonglee-115200` is EmberZnet pure Zigbee - EZSP - Standard Zigbee (NCP) firmware
Firmware Details:
* https://darkxst.github.io/silabs-firmware-builder/ 
* https://github.com/darkxst/silabs-firmware-builder?tab=readme-ov-file#pre-compiled-firmware

tootla, 2024-10-25:
> Hello 👋
> 
> I am facing the same issue on Mac OS Sequoia 15.0.1.
> 
> * ✨ Brand new Sonoff Dongle E
> * ✅ Drivers installed
> * ♻️ Rebooted multiple times
> 
> I can help on testing / QA if needed 👍
> 
> ## Debug file
> [silabs_flasher.log](https://github.com/user-attachments/files/17446811/silabs_flasher.log)
> 
> ## MacOS driver
> <img alt="" width="200" src="https://private-user-images.githubusercontent.com/1729206/378111970-1018ca2c-1d50-4643-b94f-9f9005552585.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk4NDY5NjEsIm5iZiI6MTcyOTg0NjY2MSwicGF0aCI6Ii8xNzI5MjA2LzM3ODExMTk3MC0xMDE4Y2EyYy0xZDUwLTQ2NDMtYjk0Zi05ZjkwMDU1NTI1ODUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMjVUMDg1NzQxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NGE0NTA3ZWUzYTE0MmNiZDAwMTM2M2E2NjM5NDczYjFmZjZiM2Q3ZTU5ZjQ2MWNhOWFjOTIyZDEwYjEwNWY4OSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.yMqDu1gxrc3XwcG8m5klz05-EM_zx5yYKpKrjHwCYpY">
> 
> ## Selected device
> <img alt="" width="200" src="https://private-user-images.githubusercontent.com/1729206/378112045-b9cbc676-6ff5-46c8-9613-38c94218edf1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk4NDY5NjEsIm5iZiI6MTcyOTg0NjY2MSwicGF0aCI6Ii8xNzI5MjA2LzM3ODExMjA0NS1iOWNiYzY3Ni02ZmY1LTQ2YzgtOTYxMy0zOGM5NDIxOGVkZjEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMjVUMDg1NzQxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzI1YThlMzZmNWFiNzE3NDFiZDA0NjFjMDdlOTdlODA1NDA5NTFmNDJjODczMTZlNTU5NTE5MTBkMTYwOGRhMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.63_i43j7YW2ANqCC8ddKf8_0ONyeWRyZLcVBupCFinc">
> 
> ## Developer console logs
> <img alt="" width="200" src="https://private-user-images.githubusercontent.com/1729206/378111943-4a7e8f69-cdc0-4197-96f0-51d21f4fa219.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk4NDY5NjEsIm5iZiI6MTcyOTg0NjY2MSwicGF0aCI6Ii8xNzI5MjA2LzM3ODExMTk0My00YTdlOGY2OS1jZGMwLTQxOTctOTZmMC01MWQyMWY0ZmEyMTkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMjVUMDg1NzQxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YmY0ZjRiZDM0Y2IyZTFiYzZjNGRiMmViZDM2MGMzMzBlNzBjMGEzMjM1OGI3YmUzNDU5YzgwNTM0OTdjMGYwOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.VawQXZfGOafVMIDvlU5LBHYOUUdguAP-eBjts5pjF40">

Hello @FrancoisLef , could you check the driver version of SONOFF ZBDongle-E(cp2102n) in the Mac OS?

bluemoehre, 2024-11-04:
Same issue here on MacBook Air M2. Not possible to read firmware version via Browser.

darkxst, 2024-11-05:
From the logs posted above the web flasher looks to be hanging trying to probe CPC (Multipan) firmware. Not clear though why it would fail in browser, but then work from python with universal-silabs-flasher.

I do however plan to drop support for CPC when I update web flasher for v8.0.2 soon, this will likely atleast fix that hang.

darkxst, 2024-11-05:
For now, I have disabled CPC 115200 on the test site:
https://darkxst.github.io/cp-test/

give that a try.

bluemoehre, 2024-11-05:
Looks better in Win10 (still some exceptions and errors in console); no change in MacOS.

closed 2026-01-03 by github-actions[bot]