darkxst/silabs-firmware-builder #133: SMLight SLZB-07Mg24 already supported?

stickpin, 2024-10-05:
Hi everyone,

Does anyone know if the SMLight SLZB-07Mg24, which is based on the EFR32MG24 SoC, is already supported? Should I just flash the regular SLZB-07 firmware?

Thanks a lot in advance!

darkxst, 2024-10-05:
I still need to add SLZB-07Mg24 builds here, however it ships from factory with my firmware on it.

nanostra, 2024-10-08:
Hello, I just received my SLZB-07 MG24 and trying to update firmware to 7.4.4... all seems to be ok during the process... but it stay in 7.4.2 after update !?

[image]
[image]
[silabs_flasher.log](https://github.com/user-attachments/files/17294932/silabs_flasher.log)

darkxst, 2024-10-09:
no you can't use the SLZB-07 firmware. just stick with the factory 7.4.2 firmware until I add builds here. There are no major changes in Zigbee since 7.4.2 anyway.

darkxst, 2024-10-09:
7.4.4 builds are in this branch now
https://github.com/darkxst/silabs-firmware-builder/tree/SLZB-07Mg24

Use the SLZB-07 flasher but select to upload custom firmware
https://github.com/darkxst/silabs-firmware-builder/raw/refs/heads/SLZB-07Mg24/firmware_builds/slzb-07mg24/ncp-uart-hw-v7.4.4.0-slzb-07mg24-115200.gbl

stickpin, 2024-10-09:
@darkxst thanks a lot! :) 
just flashed mine. will try to migrate now from SkyConnect to SLZB-07Mg24.

nanostra, 2024-10-09:
@darkxst That's great, thank you so much! I'm currently using a ZBDONGLE-E based on the EFR32MG21 with 85 devices. Do you think switching to an EFR32MG24 will improve OTA updates for battery-operated devices? Right now, I have to do this on a dedicated setup without any other connected devices, otherwise, it never succeeds. Regarding speed, there are always 115200 and 230400, why is 115200 generally recommended? It seems counterproductive in principle. One more question, is the flow control on this firmware hardware or software in this version? Thanks again.

darkxst, 2024-10-11:
MG24 should help with larger networks, but not sure it will change anything regarding OTA updates, but worth a try and let me know if it does.

Raw zigbee channel is 250kbps, however for Zigbee NCP firmware this is processed on the dongle and the actual messages between dongle and computer will be much much less. In most normal cases 230400 is simply not required, if you have a large network with lots of chatty devices 230400 may help, but I yet to see any concrete evidence that this is the case.

As side note, RCP firmware such as Thread or Multi-pan process zigbee traffic on the computer thus the raw zigbee 250kbps must be transferred over the serial link, which is the reason why these firmware are usually 460800 to allow for the extra bandwidth requirements.

stickpin, 2024-11-07:
Just if anyone is interested, I've installed SLZB-07MG24 v8.0.2.0 and v2.5.2 bootloader from this [20241106](https://github.com/darkxst/silabs-firmware-builder/releases/tag/20241106) release. Running it with Zigbee2MQTT, and so far so good. :)

darkxst, 2024-11-09:
SLZB-07MG24 support is complete now for both 7.4.4 and 8.0.2.

closed 2024-11-09 by darkxst