darkxst/silabs-firmware-builder #135: Connection failed

svenschedlbauer, 2024-10-16:
I have no tried so many different things, but I am unable to flash the Sonoff Dongle. I have tried it with different Dongle and put them direct on the computer and also with usb data calbe, but it will not work... I have also tested it on my PC and Laptop. Only less time I see the screen, where I can change the firmware, but when I start to flash it, it will fail.
[image]
[image]
[silabs_flasher.log](https://github.com/user-attachments/files/17401108/silabs_flasher.log)
What I am doing wrong.

darkxst, 2024-10-16:
There are currently issues with windows drivers for newer zbdongle-e stock.

Try with this test site, this fixes it for some people but not all.
https://darkxst.github.io/cp-test/

If you use Home Assistant you can also try with the "Silabs Flasher" addon

phalox, 2024-10-16:
Using the test site; it detected the firmware version installed. But the upgrade doesn`t seem to ever finish...

[image]

When checking dev tools I see this:

[image]

svenschedlbauer, 2024-10-17:
https://darkxst.github.io/cp-test/ was the solution over Chrome Browser.
First I have updated the Firmware and after I have changed the firmware to Thread.
Now it is working!!! :-) 
Thanks!!!

clement-brodu, 2024-10-22:
Hello, 
Same issue here 😞
Is there any news about this ?

phalox, 2024-10-22:
I went the linux route, installed the https://pypi.org/project/universal-silabs-flasher/ and flashed the most recent firmware to the stick.

Xen3r0, 2024-11-09:
Hello

Its doesn't work for me, I have tried with https://darkxst.github.io/silabs-firmware-builder/, but cannot connect to my dongle.

I have tried with https://darkxst.github.io/cp-test/, I can connect to my dongle, but the update or firmware flash doesn't, let's see the message on devtools (chrome) :

```
pyodide.asm.js:9 Uncaught (in promise) PythonError: asyncio.exceptions.InvalidStateError: invalid state

    at new_error (pyodide.asm.js:9:12519)
    at pyodide.asm.wasm:0x158827
    at pyodide.asm.wasm:0x15892c
    at Module.callPyObjectKwargs (pyodide.asm.js:9:81856)
    at Module.callPyObject (pyodide.asm.js:9:82066)
    at onRejected (pyodide.asm.js:9:59802)
```

```
pyodide.asm.js:9 Uncaught (in promise) PythonError: Traceback (most recent call last):
  File "/lib/python3.11/site-packages/universal_silabs_flasher/flasher.py", line 305, in enter_bootloader
    await self.probe_app_type(types=[ApplicationType.GECKO_BOOTLOADER])
  File "/lib/python3.11/site-packages/universal_silabs_flasher/flasher.py", line 261, in probe_app_type
    raise RuntimeError("Failed to probe running application type")
RuntimeError: Failed to probe running application type

    at new_error (pyodide.asm.js:9:12519)
    at pyodide.asm.wasm:0x158827
    at pyodide.asm.wasm:0x15fcd5
    at _PyCFunctionWithKeywords_TrampolineCall (pyodide.asm.js:9:123052)
    at pyodide.asm.wasm:0x1a3091
    at pyodide.asm.wasm:0x289e4d
    at pyodide.asm.wasm:0x1e3f77
    at pyodide.asm.wasm:0x1a3579
    at pyodide.asm.wasm:0x1a383a
    at pyodide.asm.wasm:0x1a38dc
    at pyodide.asm.wasm:0x2685c5
    at pyodide.asm.wasm:0x26e3d0
    at pyodide.asm.wasm:0x1a3a04
    at pyodide.asm.wasm:0x1a3694
    at pyodide.asm.wasm:0x15f45e
    at Module.callPyObjectKwargs (pyodide.asm.js:9:81732)
    at Module.callPyObject (pyodide.asm.js:9:82066)
    at wrapper (pyodide.asm.js:9:58562)
new_error @ pyodide.asm.js:9
$wrap_exception @ pyodide.asm.wasm:0x158827
$FutureDoneCallback_call @ pyodide.asm.wasm:0x15fcd5
_PyCFunctionWithKeywords_TrampolineCall @ pyodide.asm.js:9
$_PyObject_MakeTpCall @ pyodide.asm.wasm:0x1a3091
$func3387 @ pyodide.asm.wasm:0x289e4d
$func2037 @ pyodide.asm.wasm:0x1e3f77
$func1064 @ pyodide.asm.wasm:0x1a3579
$_PyObject_Call @ pyodide.asm.wasm:0x1a383a
$PyObject_Call @ pyodide.asm.wasm:0x1a38dc
$_PyEval_EvalFrameDefault @ pyodide.asm.wasm:0x2685c5
$func3201 @ pyodide.asm.wasm:0x26e3d0
$_PyFunction_Vectorcall @ pyodide.asm.wasm:0x1a3a04
$PyObject_Vectorcall @ pyodide.asm.wasm:0x1a3694
$_pyproxy_apply @ pyodide.asm.wasm:0x15f45e
Module.callPyObjectKwargs @ pyodide.asm.js:9
Module.callPyObject @ pyodide.asm.js:9
wrapper @ pyodide.asm.js:9
await in wrapper
(anonymous) @ flashing-dialog-86c94e0b.js:347
await in (anonymous)
handleEvent @ nabucasa-zigbee-flasher.js:1
```

How can i update & flash my firmware ?

Thanks

darkxst, 2024-11-09:
Please see the following comment for workarounds on windows:
https://github.com/darkxst/silabs-firmware-builder/issues/107#issuecomment-2454076415

I am going to close this issue, please comment in #107 if further issues with workarounds.

closed 2024-11-09 by darkxst