darkxst/silabs-firmware-builder #136: web-flasher and universial-silabs-flasher fail with "Failed to probe running application type"

Christian1984, 2024-10-16:
Hi! Thanks for your great work on the project!!! I tried to flash my sonoff zigbee 3.0 USB dongle plus e with a new (thread) firmware, particularly [ot-rcp-v2.4.4.0-zbdonglee-460800.gbl](https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ot-rcp-v2.4.4.0-zbdonglee-460800.gbl), which, judging from the webflasher should be the correct one!?

I'm on windows and I tried both the webflasher and the universial-silabs-flasher, in both cases I get an error:

```
Failed to probe running application type
```

Webflasher: Stays at 0.0% and dev console shows this 
[image]

universial-silabs-flasher:
[image]

Any thoughts on what I could do to get the thread firmware onto that stick? I also tried flashing with my Mac, but there it won't even let me install the silabs drivers...

darkxst, 2024-10-16:
Not sure what is causing this error, however there are known issues with the Windows drivers for cp2102 that havent been resolved yet.

Try with this command, and paste full logs if it still fails:
```
universal-silabs-flasher -v --device COM4 --bootloader-reset sonoff flash  --probe-method bootloader --firmware .\ot-rcp-v2.4.4.0-zbdonglee-460800.gbl --allow-cross-flashing
```

Christian1984, 2024-10-17:
Thanks for coming back to me, @darkxst !

Here's the log (I had to move the `--probe-method bootloader` toggle before the flash command, though):

```powershell
> universal-silabs-flasher -v --device COM4 --bootloader-reset sonoff --probe-method bootloader flash --firmware .\ot-rcp-v2.4.4.0-zbdonglee-460800.gbl --allow-cross-flashing
2024-10-17 11:40:57.984 DESKTOP-4C2SUJG asyncio DEBUG Using proactor: IocpProactor
2024-10-17 11:40:57.985 DESKTOP-4C2SUJG universal_silabs_flasher.flash INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=1, sdk_version='4.4.4', ezsp_version=None, ot_rcp_version='SL-OPENTHREAD/2.4.4.0_GitHub-7074a43e4' (2.4.4.0), cpc_version=None, fw_type=<FirmwareImageType.OT_RCP: 'ot-rcp'>, baudrate=460800)
2024-10-17 11:40:57.986 DESKTOP-4C2SUJG universal_silabs_flasher.flash DEBUG Probing with 460800 baudrate first
2024-10-17 11:40:57.986 DESKTOP-4C2SUJG universal_silabs_flasher.flasher INFO Triggering sonoff bootloader
2024-10-17 11:40:57.986 DESKTOP-4C2SUJG zigpy.serial DEBUG Opening a serial connection to 'COM4' (baudrate=115200, xonxoff=False, rtscts=False)
2024-10-17 11:40:58.217 DESKTOP-4C2SUJG universal_silabs_flasher.common DEBUG Connection made: SerialTransport(<ProactorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.common.SerialProtocol object at 0x00000272ED986E60>, Serial<id=0x272ed986e30, open=True>(port='COM4', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2024-10-17 11:40:58.225 DESKTOP-4C2SUJG universal_silabs_flasher.common DEBUG Received data b'\x11'
2024-10-17 11:40:58.349 DESKTOP-4C2SUJG universal_silabs_flasher.common DEBUG Received data b'\r\nSonoff v1.0.1\r\n\x00\r\nGecko Bootloader v1.12.00\r\n1. upload gbl\r\n2.'
2024-10-17 11:40:58.365 DESKTOP-4C2SUJG universal_silabs_flasher.common DEBUG Received data b' run\r\n3. ebl info\r\nBL > \x00'
2024-10-17 11:40:58.830 DESKTOP-4C2SUJG universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-10-17 11:40:58.830 DESKTOP-4C2SUJG zigpy.serial DEBUG Opening a serial connection to 'COM4' (baudrate=115200, xonxoff=False, rtscts=False)
2024-10-17 11:40:59.016 DESKTOP-4C2SUJG universal_silabs_flasher.common DEBUG Connection made: SerialTransport(<ProactorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.gecko_bootloader.GeckoBootloaderProtocol object at 0x00000272ED986EC0>, Serial<id=0x272ed986ef0, open=True>(port='COM4', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2024-10-17 11:40:59.016 DESKTOP-4C2SUJG universal_silabs_flasher.common DEBUG Sending data b'\n'
2024-10-17 11:40:59.017 DESKTOP-4C2SUJG universal_silabs_flasher.common DEBUG Sending data b'3'
2024-10-17 11:40:59.038 DESKTOP-4C2SUJG universal_silabs_flasher.common DEBUG Received data b'\r\nGecko Bootloader v1.12.00\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00\r\nGecko Bootloader v1.12.00\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00'
2024-10-17 11:40:59.038 DESKTOP-4C2SUJG universal_silabs_flasher.gecko_bootloader DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\r\nGecko Bootloader v1.12.00\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00\r\nGecko Bootloader v1.12.00\r\n1. upload gbl\r\n2. run\r\n3. ebl info\r\nBL > \x00')
2024-10-17 11:40:59.038 DESKTOP-4C2SUJG universal_silabs_flasher.gecko_bootloader DEBUG Detected version string '1.12.00'
2024-10-17 11:41:00.060 DESKTOP-4C2SUJG universal_silabs_flasher.flasher INFO Detected bootloader version '1.12.0'
2024-10-17 11:41:00.060 DESKTOP-4C2SUJG universal_silabs_flasher.flasher INFO Detected ApplicationType.GECKO_BOOTLOADER, version '1.12.0' at 115200 baudrate (bootloader baudrate 115200)
2024-10-17 11:41:00.060 DESKTOP-4C2SUJG universal_silabs_flasher.flash INFO Firmware baudrate 115200 differs from expected baudrate 460800
ot-rcp-v2.4.4.0-zbdonglee-460800.gbl  [------------------------------------]    0%2024-10-17 11:41:00.066 DESKTOP-4C2SUJG zigpy.serial DEBUG Opening a serial connection to 'COM4' (baudrate=115200, xonxoff=False, rtscts=False)
2024-10-17 11:41:00.374 DESKTOP-4C2SUJG universal_silabs_flasher.common DEBUG Connection made: SerialTransport(<ProactorEventLoop running=True closed=False debug=False>, <universal_silabs_flasher.gecko_bootloader.GeckoBootloaderProtocol object at 0x00000272ED9E8160>, Serial<id=0x272ed9e8250, open=True>(port='COM4', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=0, xonxoff=False, rtscts=False, dsrdtr=False))
2024-10-17 11:41:00.374 DESKTOP-4C2SUJG universal_silabs_flasher.common DEBUG Sending data b'\n'
2024-10-17 11:41:00.374 DESKTOP-4C2SUJG universal_silabs_flasher.common DEBUG Sending data b'3'

  File "C:\Python310\lib\site-packages\universal_silabs_flasher\gecko_bootloader.py", line 71, in probe
    return await self.ebl_info()
  File "C:\Python310\lib\site-packages\universal_silabs_flasher\gecko_bootloader.py", line 81, in ebl_info
    await self._state_machine.wait_for_state(State.IN_MENU)
  File "C:\Python310\lib\site-packages\universal_silabs_flasher\common.py", line 115, in wait_for_state
    return await future
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
[... cut ...]
  File "C:\Python310\lib\site-packages\universal_silabs_flasher\flasher.py", line 295, in flash_firmware
    await gecko.probe()
  File "C:\Python310\lib\site-packages\universal_silabs_flasher\gecko_bootloader.py", line 70, in probe
    async with async_timeout.timeout(PROBE_TIMEOUT):
  File "C:\Python310\lib\site-packages\async_timeout\__init__.py", line 141, in __aexit__
    self._do_exit(exc_type)
  File "C:\Python310\lib\site-packages\async_timeout\__init__.py", line 228, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
```

1nutzername, 2024-10-17:
I had exactly the same issues when trying to update my recently arrived Sonoff ZBDongle-E. The solution was to open the case, press and hold the `Boot` button pressed when connecting, and begin to flash. I'm not sure when exactly to release the button, so I held it pressed the whole process.

On my device, it needed a bit of force to unscrew the case - so when closing it after flashing, consider adding some grease to the screws and don't tighten them up that strongly.

darkxst, 2024-10-17:
> Here's the log (I had to move the `--probe-method bootloader` toggle before the flash command, though):

Both cases look like reconnect of serial causes device to reset. was hoping the probe-method command might avoid the reconnect, but seems not.

Christian1984, 2024-10-18:
@1nutzername When you say „when connecting“, do you mean physically connecting as in „when plugging the stick in“ or „when starting the process of flashing“?

darkxst, 2024-10-18:
> When you say „when connecting“, do you mean physically connecting as in „when plugging the stick in“

It should be enough to physically hold the boot button, from when you click install until the progress starts ticking over. The windows driver is resetting device right around that 0% mark.

Christian1984, 2024-10-18:
I'll give it a shot and will report back!

Christian1984, 2024-10-18:
I ended up dusting off an old, spare Raspberry Pi B and installed it with Linux. It took a while (python ran incredibly slow), but it looks like it flashed the thread firmware successfully. Thanks for your support, guys!

closed 2024-10-18 by Christian1984