darkxst/silabs-firmware-builder #138: Installation failed

MickeyILP, 2024-10-17:
Hi, I'm new to the HA family, I've heard about the many benefits of switching to ember so I wanted to try it on my dongle E. Unfortunately it turned out to be impossible on both https://darkxst.github.io/silabs-firmware-builder/ and https://darkxst.github.io/cp-test/

The first page problem is:
[image]

and the second page:
[image]

darkxst, 2024-10-18:
Hi,
  There are issues with windows drivers currently. Try using the HA Silabs Flasher addon instead. There is a hidden field that you can enter a custom URL.

MickeyILP, 2024-10-18:
> Hi, There are issues with windows drivers currently. Try using the HA Silabs Flasher addon instead. There is a hidden field that you can enter a custom URL.

What URL should i paste? https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.4.0-zbdonglee-230400.gbl - this one seems to be not working

darkxst, 2024-10-18:
You need to use the raw URL in github:
https://github.com/darkxst/silabs-firmware-builder/raw/refs/heads/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.4.0-zbdonglee-115200.gbl
https://github.com/darkxst/silabs-firmware-builder/raw/refs/heads/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.4.0-zbdonglee-230400.gbl

MickeyILP, 2024-10-18:
[image]
is it done?

darkxst, 2024-10-18:
yes, no errors so completed successfully

MickeyILP, 2024-10-18:
Everything seems to be working. Thanks a lot Sir!

closed 2024-11-09 by darkxst