darkxst/silabs-firmware-builder #139: Can't migrate from the xsp1989 for ZB-GW04 (v1.2)

ZzEeKkAa, 2024-10-21:
Any firmware from here does not work on my dongle (seems like device is bricked). However I can alway rollback to https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP/ncp-uart-sw_7.4.3.0_115200.gbl and things work again.

```bash
$ universal-silabs-flasher --device /dev/tty.usbserial-10 --probe-method=ezsp --ezsp-baudrate=115200 probe
2024-10-21 17:29:15.324 zeka-pc universal_silabs_flasher.flasher INFO Probing ApplicationType.EZSP at 115200 baud
2024-10-21 17:29:18.527 zeka-pc universal_silabs_flasher.flasher INFO Detected ApplicationType.EZSP, version '7.4.3.0 build 0' (7.4.3.0.0) at 115200 baudrate (bootloader baudrate None)
2024-10-21 17:29:18.528 zeka-pc universal_silabs_flasher.flash INFO Dumping EmberZNet Config
CONFIG_PACKET_BUFFER_COUNT=255
CONFIG_NEIGHBOR_TABLE_SIZE=26
CONFIG_APS_UNICAST_MESSAGE_COUNT=64
CONFIG_BINDING_TABLE_SIZE=32
CONFIG_ADDRESS_TABLE_SIZE=8
CONFIG_MULTICAST_TABLE_SIZE=8
CONFIG_ROUTE_TABLE_SIZE=16
CONFIG_DISCOVERY_TABLE_SIZE=8
CONFIG_STACK_PROFILE=0
CONFIG_SECURITY_LEVEL=5
CONFIG_MAX_HOPS=30
CONFIG_MAX_END_DEVICE_CHILDREN=32
CONFIG_INDIRECT_TRANSMISSION_TIMEOUT=3000
CONFIG_END_DEVICE_POLL_TIMEOUT=8
CONFIG_TX_POWER_MODE=0
CONFIG_DISABLE_RELAY=0
CONFIG_TRUST_CENTER_ADDRESS_CACHE_SIZE=0
CONFIG_SOURCE_ROUTE_TABLE_SIZE=128
CONFIG_FRAGMENT_WINDOW_SIZE=1
CONFIG_FRAGMENT_DELAY_MS=0
CONFIG_KEY_TABLE_SIZE=64
CONFIG_APS_ACK_TIMEOUT=1600
CONFIG_ACTIVE_SCAN_DURATION=3
CONFIG_PAN_ID_CONFLICT_REPORT_THRESHOLD=2
CONFIG_REQUEST_KEY_TIMEOUT=0
CONFIG_APPLICATION_ZDO_FLAGS=0
CONFIG_BROADCAST_TABLE_SIZE=128
CONFIG_MAC_FILTER_TABLE_SIZE=15
CONFIG_SUPPORTED_NETWORKS=1
CONFIG_SEND_MULTICASTS_TO_SLEEPY_ADDRESS=0
CONFIG_ZLL_GROUP_ADDRESSES=0
CONFIG_ZLL_RSSI_THRESHOLD=216
CONFIG_MTORR_FLOW_CONTROL=1
CONFIG_RETRY_QUEUE_SIZE=16
CONFIG_NEW_BROADCAST_ENTRY_THRESHOLD=122
CONFIG_TRANSIENT_KEY_TIMEOUT_S=300
CONFIG_BROADCAST_MIN_ACKS_NEEDED=255
CONFIG_TC_REJOINS_USING_WELL_KNOWN_KEY_TIMEOUT_S=300
CONFIG_CTUNE_VALUE=128
CONFIG_ASSUME_TC_CONCENTRATOR_TYPE=1
CONFIG_GP_PROXY_TABLE_SIZE=5
EZSP_CONFIG_GP_SINK_TABLE_SIZE=0
```

I've tried 
```
easyiot_zb-gw04-1v2_ncp-uart-hw_115200_8.0.2.0.gbl
ncp-uart-hw-v7.3.1.0-zb-gw04-1v2-115200.gbl
ncp-uart-hw-v7.4.3.0-zb-gw04-1v2-115200.gbl
ncp-uart-hw-v7.4.4.0-zb-gw04-1v2-115200.gbl
ncp-uart-hw-v7.4.4.0-zb-gw04-1v2-230400.gbl
rcp-uart-802154-v4.3.1-zb-gw04-1v2-230400.gbl
zb-gw04-1v2_zigbee_ncp_8.0.2.0_115200.gbl
```

But none of them worked. Also I was preflashing nvme with nvm3_initfile.gbl from xsp1989, but it also did not help.

There was no output on `screen /dev/tty.usbserial-10 115200`.

I've just got my adapter from https://www.aliexpress.us/item/3256805084701578.html . Reviews says it works both with ZHA and Zigbee2MQTT, but mine works with ZHA only.

What am I doing wrong and if there is an option to check if my hardware is malfunctioning? 

[image]

darkxst, 2024-10-21:
This device should work, v8.0.2 builds havent been tested yet, the but the others should be fine. Is the u-s-f probe from above with these firmware or xsp1989s ?

> Also I was preflashing nvme with nvm3_initfile.gbl from xsp1989, but it also did not help.

This usually fixes any issues migrating between different firmware.

ZzEeKkAa, 2024-10-22:
> This device should work, v8.0.2 builds havent been tested yet, the but the others should be fine.
I've started with 7.4.4, but since it did not work - I've been playing around with other options

> Is the u-s-f probe from above with these firmware or xsp1989s ?
From xsp1989s. Probe from these firmware fails.

> > Also I was preflashing nvme with nvm3_initfile.gbl from xsp1989, but it also did not help.
> 
> This usually fixes any issues migrating between different firmware.
I was hoping for this, once I found it in another issue, but unfortunately it did not do the trick. However the network information was erased.

Do you know any ways to debug why is it happening this way?

ZzEeKkAa, 2024-10-22:
Also I'm wondering if there is a way to play around with different versions of bootloader?

wastez, 2024-10-25:
The nvm3_init file helps. 
install the old xsp1989 firmware on it (with the flag  --bootloader-reset sonoff you should be able to do that)
Make a backup of your stick (the init file will clear your stick)
Then use the init file. 
After it install the new firmware. 
Restore your stick. 
Done.

ZzEeKkAa, 2024-10-28:
> The nvm3_init file helps. install the old xsp1989 firmware on it. Make a backup of your stick (the init file will clear your stick, if you have no devices connected or not much and repairing is an option you can forward to the next point) Then use the init file. After it install the new firmware. Restore your stick. (Of course only if you made a backup) Done.

Thank you @wastez ! I did exactly this way, but it did not work( I guess I may have malfunctioning hardware. It suppose to work with MQTT out of the box, but it does not

ZzEeKkAa, 2024-10-31:
Okay, I've figured out that I was misconfiguring mosquito, that why MQTT was not working. Now MQTT works! I was also managed to make this firmware work, but with ember-nohw branch, to precise, this build: https://github.com/darkxst/silabs-firmware-builder/blob/ember-nohw/firmware_builds/zb-gw04-1v2/ncp-uart-hw-v7.4.3.0-zb-gw04-1v2-115200.gbl

So I think I have just one issue for now - why hw does not work with gw04-1v2 ?

xsp1989, 2025-03-17:
> Okay, I've figured out that I was misconfiguring mosquito, that why MQTT was not working. Now MQTT works! I was also managed to make this firmware work, but with ember-nohw branch, to precise, this [...]

Switching in all firmwares may cause the total number of NVM tokens to exceed 200. However, GSDK sets 200 as the default value. After switching some firmwares, the total number of tokens exceeds 200, which will cause errors in the ncp initialization phase and then restart. There are two ways to solve this problem:
1. Refresh nvm_init.gbl, clear all nvm data, and after flashing the new firmware, the number of tokens will be less than 200, and your program can run normally.
2. Modify the value of the firmware's nvm cache, change it to 210 or above, such as 250, which will avoid this problem.
See the discussion here

https://github.com/xsp1989/zigbeeFirmware/issues/28

closed 2026-01-03 by github-actions[bot]