darkxst/silabs-firmware-builder #142: I cant flash my Sonoff ZBDongle with web flasher

R4fte, 2024-11-03:
Hey,
I can't flash my Sonoff key with the browser, even though I followed the steps by steps. My key is recognized by the browser, but it tells me that I can't connect and connection failed

The image when we told me to choice a device : 
[image]
The image when its failed : [image]
The Debug log : [silabs_flasher.log](https://github.com/user-attachments/files/17610542/silabs_flasher.log)

What can I do? Thanks in advance for your answer

tootla, 2024-11-04:
https://github.com/darkxst/silabs-firmware-builder/issues/107#issuecomment-2436970961

tootla, 2024-11-04:
@R4fte Hello guy, maybe you can try changing the Windows driver of CP210X , like this : : [https://github.com/darkxst/silabs-firmware-builder/issues/107#issuecomment-2436970961](https://github.com/darkxst/silabs-firmware-builder/issues/107#issuecomment-2436970961)

darkxst, 2024-11-04:
Closing as this is well covered in #107

closed 2024-11-04 by darkxst

R4fte, 2024-11-06:
Hi back, I Install the driver who u told me but nothing happend, idk what I can do more ?
[silabs_flasher.log](https://github.com/user-attachments/files/17647528/silabs_flasher.log)
[image]
[image]
[image]
I follow all step who's written