darkxst/silabs-firmware-builder #143: Bricked my ZB-GW04 v1.2 ZYZBP008 🧱

newzealandpaul, 2024-11-04:
I flashed the wrong firmware (slzb07Mg24_bootloader_2.5.2.gbl) by mistake 😭 Firmware upload went fine but the device can no longer be detected.

```
$ universal-silabs-flasher --device /dev/cu.usbserial-210 --probe-method=ezsp --ezsp-baudrate=115200 probe
2024-11-04 18:08:14.257 Mac.15.nz universal_silabs_flasher.flasher INFO Probing ApplicationType.EZSP at 115200 baud
Error: Failed to probe running application type
```

Probe info from before the bad flash:

```
2024-11-04 17:48:04.131 x universal_silabs_flasher.flasher INFO Detected ApplicationType.EZSP, version '6.10.3.0 build 297' (6.10.3.0.297) at 115200 baudrate (bootloader baudrate None)
2024-11-04 17:48:04.131 x universal_silabs_flasher.flash INFO Dumping EmberZNet Config
2024-11-04 17:48:04.133 x zigpy.serial DEBUG Opening a serial connection to '/dev/cu.usbserial-210' (baudrate=115200, xonxoff=True, rtscts=False) 

...

CONFIG_PACKET_BUFFER_COUNT=200
CONFIG_NEIGHBOR_TABLE_SIZE=26
CONFIG_APS_UNICAST_MESSAGE_COUNT=128
CONFIG_BINDING_TABLE_SIZE=32
CONFIG_ADDRESS_TABLE_SIZE=200
CONFIG_MULTICAST_TABLE_SIZE=8
CONFIG_ROUTE_TABLE_SIZE=16
CONFIG_DISCOVERY_TABLE_SIZE=8
CONFIG_STACK_PROFILE=0
CONFIG_SECURITY_LEVEL=5
CONFIG_MAX_HOPS=30
CONFIG_MAX_END_DEVICE_CHILDREN=32
CONFIG_INDIRECT_TRANSMISSION_TIMEOUT=3000
CONFIG_END_DEVICE_POLL_TIMEOUT=8
CONFIG_TX_POWER_MODE=0
CONFIG_DISABLE_RELAY=0
CONFIG_TRUST_CENTER_ADDRESS_CACHE_SIZE=0
CONFIG_SOURCE_ROUTE_TABLE_SIZE=128
CONFIG_FRAGMENT_WINDOW_SIZE=1
CONFIG_FRAGMENT_DELAY_MS=0
CONFIG_KEY_TABLE_SIZE=12
CONFIG_APS_ACK_TIMEOUT=1600
CONFIG_ACTIVE_SCAN_DURATION=3
CONFIG_END_DEVICE_BIND_TIMEOUT=60
CONFIG_PAN_ID_CONFLICT_REPORT_THRESHOLD=2
CONFIG_REQUEST_KEY_TIMEOUT=0
CONFIG_CERTIFICATE_TABLE_SIZE=0
CONFIG_APPLICATION_ZDO_FLAGS=0
CONFIG_BROADCAST_TABLE_SIZE=128
CONFIG_MAC_FILTER_TABLE_SIZE=0
CONFIG_SUPPORTED_NETWORKS=1
CONFIG_SEND_MULTICASTS_TO_SLEEPY_ADDRESS=0
CONFIG_ZLL_GROUP_ADDRESSES=1
CONFIG_ZLL_RSSI_THRESHOLD=128
CONFIG_MTORR_FLOW_CONTROL=1
CONFIG_RETRY_QUEUE_SIZE=16
CONFIG_NEW_BROADCAST_ENTRY_THRESHOLD=122
CONFIG_TRANSIENT_KEY_TIMEOUT_S=300
CONFIG_BROADCAST_MIN_ACKS_NEEDED=255
CONFIG_TC_REJOINS_USING_WELL_KNOWN_KEY_TIMEOUT_S=300
CONFIG_CTUNE_VALUE=128
CONFIG_ASSUME_TC_CONCENTRATOR_TYPE=1
```

Is it totally bricked or is there some hope?

darkxst, 2024-11-04:
You can now only flash new bootloader using SWD or JTAG. I dont think these pins are broken out on zb-gw04 board, so you would need to attach directly to the pins on the zigbee module. I suppose a suitable debugger for this is going to cost more than a new zb-gw04... or take the opportunity to buy a better dongle ;)

newzealandpaul, 2024-11-04:
@darkxst thanks for confirming. It was a pretty awful dongle.

closed 2024-11-04 by newzealandpaul