darkxst/silabs-firmware-builder #145: mayb I broke my Zigbee 3.0 USB Dongle Plus

IngiDingi, 2024-11-10:
Hi.

Today I tried to use the https://darkxst.github.io/silabs-firmware-builder/ to flash my Zigbee 3.0 USB Dongle Plus to the MulitPAN (RCP).
The deice was recongnized by the browser in COM 6 and I allowed connection.
After choosing the firmware I started the process.
But it was stuck at 0% for 10min so I decidec to cancel this process and start from scratch on another PC.

This is not the issue:
[image]
Here is the LOG:
[silabs_flasher.log](https://github.com/user-attachments/files/17690959/silabs_flasher.log)

I tried to recover the original firmware useing SecureCRT but I can not connect to the Zigbee 3.0 USB Dongle Plus at all.
Activating the bootlader in itself is not even possible anymore I guess.
I tried this method:

darkxst, 2024-11-10:
The only way to brick dongle is to flash incorrect bootloader, in all other cases you should be able to recover using the internal boot button.

tootla, 2024-11-11:
I agreed with darkxst. I thought that ZBDongle didn't enter the bootloader when you manually flashed the firmware, because after entering the bootloader, the menu should have appeared on SecureCRT.

IngiDingi, 2024-11-11:
Hi. Thx for your reply. Today I tried with my old Laptop and with this I was able to flash again. Something which is similar on my PCs is blocking the communication.