darkxst/silabs-firmware-builder #149: OTA problem

antst, 2024-11-21:
When I am trying to do OTA firmware updates, I always end up with this error
```
[2024-11-21 17:20:23] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-11-21 17:20:23] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2024-11-21 17:20:23] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-11-21 17:20:23] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Total frames: RX=1699, TX=2135
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   DATA frames : RX=1694, TX=433
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   DATA bytes  : RX=52446, TX=24903
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Retry frames: RX=3, TX=3
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=1698
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Retry dupes     : RX=3
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-11-21 17:20:23] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=3
[2024-11-21 17:20:32] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=175 Len=10].
[2024-11-21 17:20:54] info: 	zh:ember:uart:ash: Port closed.
[2024-11-21 17:20:54] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2024-11-21 17:20:54] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2024-11-21 17:20:54] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2024-11-21 17:20:54] error: 	z2m: Adapter disconnected, stopping
```

I am using Z2M, but it does not look like Z2M problem (correct me if I am wrong). This log is from 8.02, but same story with 7.4.4
It just matter of time till it crashes with this error. Sometimes I can get through 3% of update, sometimes through 7%...but end is always the same.

I have relatively dense network with 110 devices.

akaSyntaax, 2024-11-30:
I can confirm the same behavior for 7.4.3, 7.4.4 and 7.4.5 on a Sonoff ZBDongle-E V2 using Zigbee2MQTT 1.41.0. The error is as follows:

`[2024-11-30 16:28:23] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-11-30 16:28:23] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2024-11-30 16:28:23] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-11-30 16:28:23] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2024-11-30 16:28:23] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2024-11-30 16:28:23] info: 	zh:ember:uart:ash:   Total frames: RX=80, TX=128
[2024-11-30 16:28:23] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2024-11-30 16:28:23] info: 	zh:ember:uart:ash:   DATA frames : RX=70, TX=46
[2024-11-30 16:28:23] info: 	zh:ember:uart:ash:   DATA bytes  : RX=999, TX=585
[2024-11-30 16:28:23] info: 	zh:ember:uart:ash:   Retry frames: RX=8, TX=2
[2024-11-30 16:28:23] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=79
[2024-11-30 16:28:23] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2024-11-30 16:28:23] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2024-11-30 16:28:23] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2024-11-30 16:28:23] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2024-11-30 16:28:23] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2024-11-30 16:28:23] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2024-11-30 16:28:23] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2024-11-30 16:28:23] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2024-11-30 16:28:23] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2024-11-30 16:28:23] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2024-11-30 16:28:23] info: 	zh:ember:uart:ash:   Retry dupes     : RX=8
[2024-11-30 16:28:23] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2024-11-30 16:28:23] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=2
[2024-11-30 16:28:31] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=44 Len=7].
[2024-11-30 16:28:53] info: 	zh:ember:uart:ash: Port closed.
[2024-11-30 16:28:53] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2024-11-30 16:28:53] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2024-11-30 16:28:53] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2024-11-30 16:28:53] error: 	z2m: Adapter disconnected, stopping`

I tried devices from Aqara and Bosch.

antst, 2024-12-02:
this is, not necessary firmware error (even if Z2M manages to do it).
I tried same devices in small network of 5 devices, and it all worked, even simultaneous update of 5 devices.
There must be something off with one of devices in the network. I think

tobzsc, 2024-12-30:
I can also confirm this error. I am now on 8.0.2 but also had this problem with 7.4.4. 

@antst I do not think this is a faulty device. The protocol stack should be solid enough to handle faulty devices and not quit operation. I have 10 devices in my network, so I am also pretty small.

tootla, 2025-01-14:
@darkxst Here is the latest progress on the `ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT` error issue:
[https://github.com/Koenkk/zigbee2mqtt/issues/23861#issuecomment-2581927487](https://github.com/Koenkk/zigbee2mqtt/issues/23861#issuecomment-2581927487). 

It currently appears that adjusting the firmware's `SL_IOSTREAM_USART_VCOM_RX_BUFFER_SIZE` to `128` can alleviate this problem, and more users are still testing it.

darkxst, 2025-01-16:
I have applied this change and built test firmware, please test and report back results.

https://github.com/darkxst/silabs-firmware-builder/actions/runs/12819462115