darkxst/silabs-firmware-builder #155: FR: ZB-GW04 v1.2 router builds

erkr, 2024-11-24:
Hi 

I’m searching for a (recent) router build for the ZB-GW04 v1.2 dongles sold on Ali. 
Any reason I don’t see them listed in the firmware folder?

Best Eric

darkxst, 2024-11-24:
I havent previously built router builds, however they will be coming in the next few weeks.

In the interim Sonoff router builds should work on that dongle
https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/blob/master/Dongle-E/Router/Z3RouterUSBDonlge_EZNet6.10.3_V1.0.0.gbl

erkr, 2024-11-24:
Thanks for the quick reply 🤗

DavidAntonin, 2025-01-22:
@darkxst I am interested too :)

erkr, 2025-01-22:
Yes still waiting patiently ☺️

Nerivec, 2025-03-15:
https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2-update6
_I haven't gotten any [feedback](https://github.com/Nerivec/silabs-firmware-builder/discussions/16) on this model though..._

erkr, 2025-03-15:
> https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2-update6

Thanks. That link is new to me. Will try that build and report back direct after install and pair. 
To judge stability, I have to test  over a significant period

Nerivec, 2025-03-19:
Thanks for the feedback.

https://www.zigbee2mqtt.io/guide/adapters/emberznet.html#recommended
Pick one from the MG24 series. They have far more RAM than the MG21 series. _That link has links to the chip used for each adapter, with the full specs, if you're curious._
_Same applies for coordinators by the way._

erkr, 2025-03-19:
For the coördinator I fully agree, but for routers these cheap dongles will do a beter then smart Plugs

erkr, 2025-04-15:
@Nerivec update: your router build runs very well. Thanks

closed 2026-01-01 by github-actions[bot]