darkxst/silabs-firmware-builder #156: problems with webflasher - debug from chrome F12 tab.

JacekKac, 2024-11-24:
Tried with different usb serial drivers, and with edge browser problem is the same. 

Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=00001800-0000-1000-8000-00805f9b34fb, name='Яндекс Станция Макс'
silabs-firmware-builder/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000110b-0000-1000-8000-00805f9b34fb, name='Яндекс Станция Макс'
pyodide.asm.js:9 Loading micropip, packaging
pyodide.asm.js:9 Loaded packaging, micropip
pyodide.asm.js:9 Uncaught (in promise) PythonError: Traceback (most recent call last):
  File "/lib/python3.11/site-packages/packaging/requirements.py", line 35, in __init__
    parsed = _parse_requirement(requirement_string)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/lib/python3.11/site-packages/packaging/_parser.py", line 64, in parse_requirement
    return _parse_requirement(Tokenizer(source, rules=DEFAULT_RULES))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/lib/python3.11/site-packages/packaging/_parser.py", line 82, in _parse_requirement
    url, specifier, marker = _parse_requirement_details(tokenizer)
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/lib/python3.11/site-packages/packaging/_parser.py", line 126, in _parse_requirement_details
    marker = _parse_requirement_marker(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/lib/python3.11/site-packages/packaging/_parser.py", line 147, in _parse_requirement_marker
    tokenizer.raise_syntax_error(
  File "/lib/python3.11/site-packages/packaging/_tokenizer.py", line 165, in raise_syntax_error
    raise ParserSyntaxError(
packaging._tokenizer.ParserSyntaxError: Expected end or semicolon (after version specifier)
    aiohappyeyeballs==2.4.3
aiohttp
                    ~~~~~~~^

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/lib/python3.11/site-packages/micropip/_commands/install.py", line 142, in install
    await transaction.gather_requirements(requirements)
  File "/lib/python3.11/site-packages/micropip/transaction.py", line 204, in gather_requirements
    await asyncio.gather(*requirement_promises)
  File "/lib/python3.11/site-packages/micropip/transaction.py", line 211, in add_requirement
    return await self.add_requirement_inner(Requirement(req))
                                            ^^^^^^^^^^^^^^^^
  File "/lib/python3.11/site-packages/packaging/requirements.py", line 37, in __init__
    raise InvalidRequirement(str(e)) from e
packaging.requirements.InvalidRequirement: Expected end or semicolon (after version specifier)
    aiohappyeyeballs==2.4.3
aiohttp
                    ~~~~~~~^

    at new_error (pyodide.asm.js:9:12519)
    at pyodide.asm.wasm:0x158827
    at pyodide.asm.wasm:0x15fcd5
    at _PyCFunctionWithKeywords_TrampolineCall (pyodide.asm.js:9:123052)
    at pyodide.asm.wasm:0x1a3091
    at pyodide.asm.wasm:0x289e4d
    at pyodide.asm.wasm:0x1e3f77
    at pyodide.asm.wasm:0x1a3579
    at pyodide.asm.wasm:0x1a383a
    at pyodide.asm.wasm:0x1a38dc
    at pyodide.asm.wasm:0x2685c5
    at pyodide.asm.wasm:0x26e3d0
    at pyodide.asm.wasm:0x1a3a04
    at pyodide.asm.wasm:0x1a3694
    at pyodide.asm.wasm:0x15f45e
    at Module.callPyObjectKwargs (pyodide.asm.js:9:81732)
    at Module.callPyObject (pyodide.asm.js:9:82066)
    at wrapper (pyodide.asm.js:9:58562)Understand this errorAI

darkxst, 2024-11-24:
I updated the webflasher yesterday, this will be a bug in that. This is on Windows?

darkxst, 2024-11-25:
This should be fixed now.

closed 2024-11-25 by darkxst

JacekKac, 2024-12-02:
new try today:).... 

[silabs_flasher (1).log](https://github.com/user-attachments/files/17983663/silabs_flasher.1.log)

darkxst, 2024-12-03:
> 2024-12-02 17:59:29 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.CPC at 115200 baud

That is an unrelated error, not sure why it would hang on CPC (Multipan). what firmware is installed on your dongle?

JacekKac, 2024-12-03:
not sure exactly but there is a router firmware 
Z3RouterUSBDonlge_EZNet6.10.3_V1.0.0.gbl or @Nerivec
https://github.com/Nerivec/silabs-firmware-builder/releases

JacekKac, 2024-12-10:
@darkxst should i open a new ticket or is it possible to find a solution here?

darkxst, 2024-12-11:
@JacekKac If you have router firmware installed web flasher cannot reset into bootloader, you have 2 opions:

* Use the pyhsical boot buttons to enter bootloader (hold button while webflasher loads and until it starts flashing)
* Use `universal-silabs-flasher` CLI with the `--bootloader-reset sonoff` option