darkxst/silabs-firmware-builder #161: Feature request: router firmware for ZBDongle-E

jorisvervuurt, 2024-12-07:
I noticed that recently router firmware was added for the SLZB-07. 
I have a ZBDongle-E and would be very happy if a router firmware for my dongle would be added as well. :) The official one from Sonoff is over 2 years old...

darkxst, 2024-12-22:
SMLIGHT sponsored these router builds, so for the time being they are exclusive to SLZB devices. If you want to try them out purchase a SLZB-07, and support a company that actively supports the HA/Z2M ecosystem unlike Sonoff!

jorisvervuurt, 2024-12-24:
I actually moved from an SLZB-06 (the TI-based one) PoE coordinator to the Sonoff ZBDongle-E USB-based dongle due to issues with the serial connection of the SZLB-06. The Sonoff one seems to be much more stable with better range. The SLZB-07 USB one probably performs equally well, but I have now moved over to Sonoff and am not going to change that. I read mixed stuff regarding the SLZB-07 bootloader and the firmware does not seem to be open-source (it is for the Sonoff). The Sonoff does not have hardware flow control but the SLZB-06 didn't either I believe. I like the metal enclosure of the Sonoff.

I have just moved everything over to the ZBDongle-E (coordinator and two routers), so unfortunately I'm not going to change that now. I still hope one day new router builds will be available for the ZBDongle-E (even though the Sonoff one that's two years old seems to work fine).

darkxst, 2024-12-24:
> after a router/firewall reboot
This is likely something in the ESP32 firmware messing things up. 

> regarding the SLZB-07 bootloader.

An early batch of SLZB-07 had locked bootloader by accident. A fix for this was released (available in my webflasher) but all devices come with unlocked bootloader.

SLZB-07 and Dongle-E are very similar with same serial chips and zigbee chips. Hardware flow control doesnt seem to make much difference especially with Zigbee.

jorisvervuurt, 2024-12-25:
Thanks @darkxst. Out of curiosity, I have ordered a few SLZB-07 MG24 dongles to try out with the coordinator and router firmware.

Do your coordinator and router firmware builds for the SLZB-07 MG24 both have HW flow control enabled or should I use the SMLIGHT web flasher for that?

darkxst, 2024-12-25:
> Do your coordinator and router firmware builds for the SLZB-07 MG24 both have HW flow control enabled or should I use the SMLIGHT web flasher for that?

Coordinator has hardware flow control enabled, router doesnt since it doesnt use serial connection in the same way.

You can use either web flasher, should be the same builds for 7.4.x, except I dont think 8.0.2 or router builds have made it onto smlight web flasher yet. However they will  become available there shortly.

jorisvervuurt, 2024-12-25:
Ah yes, of course. Cool; thanks!

closed 2024-12-25 by jorisvervuurt