darkxst/silabs-firmware-builder #164: Z2M won't start with Dongle E and 7.4.5

CaesarNaykid, 2024-12-11:
[2024-12-11 09:21:52] info: 	z2m: Logging to console, file (filename: log.log)
[2024-12-11 09:21:52] info: 	z2m: Starting Zigbee2MQTT version 1.42.0 (commit #unknown)
[2024-12-11 09:21:52] info: 	z2m: Starting zigbee-herdsman (2.1.9)
[2024-12-11 09:21:53] info: 	zh:ember: Using default stack config.
[2024-12-11 09:21:53] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-12-11 09:21:53] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-12-11 09:21:53] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-12-11 09:21:53] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-12-11 09:21:53] info: 	zh:ember:uart:ash: Serial port opened
[2024-12-11 09:21:53] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-12-11 09:21:55] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-12-11 09:21:55] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-12-11 09:21:58] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-12-11 09:21:58] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-12-11 09:22:00] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-12-11 09:22:00] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-12-11 09:22:03] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-12-11 09:22:03] info: 	zh:ember:uart:ash: ======== ASH starting ========[2024-12-11 09:22:03] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-12-11 09:22:05] error: 	z2m: Error while starting zigbee-herdsman
[2024-12-11 09:22:05] error: 	z2m: Failed to start zigbee
[2024-12-11 09:22:05] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-12-11 09:22:05] error: 	z2m: Exiting...
[2024-12-11 09:22:05] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.

flashed to the newest: https://github.com/darkxst/silabs-firmware-builder/blob/fb33727cf639a869d14769e5c28e56bfaf927a52/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.5.0-zbdonglee-230400.gbl

any suggestions?

darkxst, 2024-12-22:
I will look into issues on 7.4.5 after xmas period.

CaesarNaykid, 2024-12-22:
I got my dongle-e to work with flashing different firmwares and baud rates, can't recall which but i think rolled back to 7.4.4 

lmk if there's any way to check from Home assistant/vm console and i'll check which one i installed

darkxst, 2024-12-22:
In ZHA just find the device for coordinator and it will show you the version.

7.4.5 was a relatively big update (compared to previous 7.4.3 and 7.4.4) perhaps they broke something in the process.

CaesarNaykid, 2024-12-23:
my apologies. I just checked in Zigbee2MQTT Settings-> About tab and it says:
Coordinator type: EmberZNet
Coordinator revision: 7.4.5 [GA]

So, maybe i only switched from the 2x to the 1x baud rate version if that's accurate from Z2M to get it to work

joeybab3, 2025-01-07:
7.4.5 and 7.4.4 have this issue for me. 7.4.3 works fine.

closed 2025-12-31 by github-actions[bot]