darkxst/silabs-firmware-builder #166: ZBDongle-E won't work after Update to 7.4.5

Viper780, 2024-12-22:
I've updated my Dongle-E from 7.4.3 to NCP 7.4.5 and ZHA brings up an error

[image]

going back to [ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl](https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl) or flashing zbdonglee_zigbee_ncp_8.0.2.0_sw_flow_115200.gbl worked.

darkxst, 2024-12-22:
I am aware of some issues with this build. However wont be able to look into until after xmas.

That message is very odd if you are sure you installed https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.5.0-zbdonglee-115200.gbl and not the corresponding RCP build.

Viper780, 2024-12-23:
Thanks for the fast reply.

Yes I'm sure (tripple checked) that I've used the NCP build.
Right now I'm on 8.0.2 and it looks stable, so sticking with the latest stack

KC-inDomus, 2025-01-07:
> Thanks for the fast reply.
> 
> Yes I'm sure (tripple checked) that I've used the NCP build. Right now I'm on 8.0.2 and it looks stable, so sticking with the latest stack

is 8.0.2 with hw flow control available? can't find it.

darkxst, 2025-01-07:
> is 8.0.2 with hw flow control available? can't find it.

not for Zbdongle-E it doesnt support hw flow control.

maxromanovsky, 2025-02-06:
@darkxst out of curiosity, why `7.x` builds were named `hw` for this dongle? Was it just naming issue?

See this diff: https://github.com/darkxst/silabs-firmware-builder/commit/9ae16bec538ea456f7dc677a5719293e2bd54956#diff-6d692c8d1f2a0fa4c91cf083a49eecdbb32c11a5b774ef8c594de996e9ffff28

darkxst, 2025-02-06:
> why `7.x` builds were named `hw` for this dongle?

Filenames were generated from the project name "ncp_uart_hw" for the old 7.4.x builder.

maxromanovsky, 2025-02-07:
Sorry for hijacking this discussion, but also what's the reason for rolling back github pages for most of the dongles to v7?
I have just updated to v8, and now I'm a bit concerned :)
https://github.com/darkxst/silabs-firmware-builder/commit/a650408761535d4542b73804cd9885c7c417af2e#diff-6d692c8d1f2a0fa4c91cf083a49eecdbb32c11a5b774ef8c594de996e9ffff28

darkxst, 2025-02-07:
there was a possible issue with mg24 firmware, no issues with the zbdongle-e firmware. I will re-update the webflasher to v8.0.2 shortly.

auanasgheps, 2025-05-27:
Hi, an update on this? The latest available on the web flasher is still 7.4.4.0.