darkxst/silabs-firmware-builder #168: SLZB-07 MG 24 router firmware not automatically reconnecting

jorisvervuurt, 2024-12-29:
@darkxst Based on your recommendation, I ordered some SMLIGHT SLZB-07 (MG24) dongles to experiment with.

I have just flashed one of the SMLIGHT SLZB-07 MG24 dongles with the router firmware (8.0.2.0).
The coordinator is also an SMLIGHT SLZB-07 MG24 running the latest 8.0.2.0 firmware.

Before flashing the router firmware, I allowed pairing in Zigbee2MQTT (2.0 dev branch since the stable branch does not support the SLZB-07) and the dongle correctly joined the network after it finished flashing the router firmware. I then unplugged the router dongle and inserted it in the 230V -> USB adapter which I previously used with a Sonoff router dongle. The SLZB-07 never joins the network again and the green LED stays off.

Why doesn't the dongle join the network again?

Nerivec, 2024-12-29:
Is it just an LED problem, or is the adapter actually not rejoining the network? Since it's a very passive device, it might show as offline until it is pinged by Z2M (either via availability, or triggered manually via network map, read in dev console...).

darkxst, 2024-12-30:
Once its paried, It should reconnect as soon as power on.
 
@jorisvervuurt  connect to the the serial console on the router and see if you can capture any logs. There are also commands you can run in the console to check network status etc, I think "info" will display network information. "network leave" will force router to re-enter pairing mode etc.

LED logic could be buggy still, initially it shows the commissioning/joining activity, but once paired the "Light entity" should override this allowing to turn the led off if you want.

Coordinator firmware doesnt have LED configured yet.

jorisvervuurt, 2024-12-30:
Thanks for the quick reply @darkxst. It never joins the network again after being paired and power is lost (other Zigbee devices work fine). I will try checking the serial console later today.

jorisvervuurt, 2024-12-30:
@Nerivec @darkxst I've done some more testing and have found the issue:
The dongle flashed with 8.0.2.0 router firmware only reconnects to the Zigbee network once a serial connection is opened.
I can image that might work for a coordinator (though I can imagine this is potentially problematic too), but not for a router dongle.

Here's the steps to reproduce and some screenshots:

1. Flash SLZB-07MG24 with 8.0.2.0 router firmware (while Zigbee network allows device joining).
2. Wait for device to join the network; green LED will illuminate.
3. Perform a network scan (Map) in Zigbee2MQTT (2.0 dev branch) and confirm LQI values are read.
4. Unplug the dongle from the computer.
5. Replug the dongle in the computer; green LED will not illuminate.
6. Again perform a network scan (Map) in Zigbee2MQTT; an error will be shown (LQI reading failed).
7. Open serial connection (e.g. using PuTTY); notice green LED shortly illuminates (serial console shows watchdog reset?).
8. Device will now be connected to the Zigbee network; confirm using network scan (Map) in Zigbee2MQTT that it is online and LQI values are read (no errors).
9. Entering `reset` command in terminal works fine too; device reconnects to Zigbee network.
10. Close serial connection, unplug dongle and replug it without opening serial connection again -> will not reconnect to Zigbee network and green LED does not shortly illuminate.

So basically, the dongle joins the network once a serial connection is opened. Of course, this cannot be done when the dongle is inserted in a USB wall charger...

Here's a screenshot just after opening the serial connection using PuTTY:
[image]

And here's a screenshot after issuing `reset` and `info` commands using PuTTY:
[image]

This seems like a firmware bug in the router firmware (a bit like a while loop in Arduino which waits for USB serial connection). ;-) Thanks again in advance!

darkxst, 2024-12-30:
@jorisvervuurt Thanks for the extra info. I will take a look and see what is going on.

> The dongle flashed with 8.0.2.0 router firmware only reconnects to the Zigbee network once a serial connection is opened. I can image that might work for a coordinator, but not for a router dongle.

Yeh the router shouldn't require a serial connection to work

jorisvervuurt, 2024-12-30:
Awesome, thanks very much. Feel free to let me know if you want me to test new firmware. ;-)

jorisvervuurt, 2024-12-30:
@Nerivec Awesome; that one works perfectly (including the LED)! I flashed the GBL using the @darkxst web flasher ( https://darkxst.github.io/silabs-firmware-builder/ ), however the web flasher now no longer recognizes the firmware. I entered bootloader mode (by shorting the two pads on the PCB) and then the web flasher showed that my dongle was running bootloader 2.4. I noticed that you provide a 2.5.2 bootloader in your repo (what's the difference?) so I flashed that as well using the web flasher.

Now there is only one problem: when I put the dongle in bootloader mode (via the reset button in Z2M), the web flasher correctly shows bootloader version 2.5.2. If I however try to reflash the bootloader (reflashing the 2.5.2 bootloader), it always says it failed. Is that normal / expected?

jorisvervuurt, 2024-12-30:
I only just noticed you also have a 3.0 bootloader firmware ( https://github.com/Nerivec/silabs-firmware-builder/actions/runs/12538101751 ) available. Is that safe to flash? And what are the differences? :)

Nerivec, 2024-12-30:
8.1.0 is the latest release from Silabs, still vastly untested, but I've tested it myself on SLZB07MG24, so I know it works on that one 😉 3.0.0 bootloader is the accompanying bootloader update, which I have **not** tested.
The bootloader is updated with every new SDK update by Silabs, so it's provided (by darkxst also). You'd have to check the [Gecko Platform release notes](https://github.com/SiliconLabs/simplicity_sdk/releases/tag/v2024.12.0) to get more details (usually very few changes though).

You indeed have to put the router firmware in bootloader manually as the web flasher uses the coordinator stack to enter the bootloader otherwise.

Commands are disabled in the router builds from my repo, only the log outputs from the firmware are present. I've removed a lot of stuff that is not useful in every-day use for a router. There are other differences with darkxst router firmware too (like the reset button). But the base functionality (routing) remains the same, rest is just "extras".

jorisvervuurt, 2024-12-30:
Thanks! As for the commands not available in your firmware: how can I reset the dongle if it is force deleted from Z2M? Normally this could be done using the `network leave` command. ;-)

I'll test the 3.0 bootloader on one of my experimental dongles just to test. ;-)

Nerivec, 2024-12-30:
Best is to avoid force delete, since it only deletes the device from Z2M database, not from the actual network, it should really be a last resort option (device no longer working).
On supporting adapters, you can do that with [the button](https://github.com/Nerivec/silabs-firmware-builder/discussions/16#discussion-7442147) but the slzb doesn't have one... so, for now, you'd have to reset the nvm3 https://github.com/Nerivec/silabs-firmware-recovery/blob/main/gbl-builds/EFR32MG24A020F1024IM40_nvm3_clear_134217728_1048576_8192_134242304_32768.gbl

_I'm looking into a way to add this without having to re-add the entire commands component and without relying solely on button (since some manuf still don't put one)._

darkxst, 2024-12-30:
Delete from Z2M should trigger re-pairing on router (atleast on my builds).

Nerivec, 2024-12-31:
@darkxst Not "force delete" (that doesn't actually do anything on the device, it just removes it from the Z2M database).

jorisvervuurt, 2024-12-31:
Hi @Nerivec and @darkxst, just wanted to let you guys know that the network with the 3.0 bootloader and 8.1.0.0 firmwares (coordinator and router) is still running stable. :)

If there is one feature request I may add: the green LED is very bright and I am getting complaints from the people I live with because some of the dongles are in their rooms. Is there a way to disable the LED once the dongle has joined the network? 😃

darkxst, 2024-12-31:
> If there is one feature request I may add: the green LED is very bright and I am getting complaints from the people I live with because some of the dongles are in their rooms. Is there a way to disable the LED once the dongle has joined the network? 😃

The light entity on my firmware allows you to disable this LED (once device is paired), not sure if this is exposed in the Z2M convertor yet though?

jorisvervuurt, 2024-12-31:
@darkxst I did see that indeed, but this is not available in @Nerivec's converter (and perhaps also not in the firmware). I only see the reset button and LQI in Z2M.

darkxst, 2024-12-31:
> I did see that indeed, but this is not available in @Nerivec's converter (and perhaps also not in the firmware).

Well it will be in the official firmware once its ready to be released. I dont think there will be an official release with the combined firmware as its not really user-friendly to switch modes. Where as switching with the web-flasher is pretty easy.

jorisvervuurt, 2024-12-31:
👍 I'll be patient. 😄 Happy New Year!

Nerivec, 2024-12-31:
_Some adapters also have a power-on LED, so the on/off support is moot in those cases._
Tape is your friend 😁

@darkxst indeed the "official" Z2M converter will need an update if you want to support on/off https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/smlight.ts

Happy new year 🥳

jorisvervuurt, 2024-12-31:
@Nerivec Yeah I also noticed the smlight.ts converter doesn't show the correct power method. The Silabs router converter you added seems to be great in my opinion. 😄

darkxst, 2025-01-01:
@jorisvervuurt You can try test this build:
https://github.com/darkxst/silabs-firmware-builder/actions/runs/12565415438

> indeed the "official" Z2M converter will need an update if you want to support on/off

Yes will do, before official release of router builds.

jorisvervuurt, 2025-01-01:
Thanks @darkxst; this build indeed correctly reconnects to the Zigbee network once the dongle is powered. :)

I did notice a few things though:
- I previously flashed the 3.0 bootloader; since yours is still at 2.5.2, I tried downgrading. It said it failed and keeps reporting 3.0. Tried reflashing 3.0 but that failed as well. Is it expected that downgrading the bootloader does not work and reflashing the same bootloader also fails (reflashing 2.5.2 on a dongle with same 2.5.2 bootloader also fails)?

- Re-paired the dongle with Z2M (deleted it and re-added it): it now uses the SMLIGHT converter. The 'Exposes' tab only shows LQI and the 'Configuring' step of the pairing process fails with this error (saw this with your earlier router firmwares as well - @Nerivec's firmware does not show this error during the 'Configuring' step):
```
z2m: Failed to configure '0x90395efffecda221', attempt 3 (Error: ZCL command 0x90395efffecda221/1 genBasic.configReport([{"minimumReportInterval":3600,"maximumReportInterval":14400,"reportableChange":0,"attribute":"modelId"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":39503,"apsFrame":{"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":21},"zclSequence":74,"commandIdentifier":7} timed out after 10000ms) at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.0/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:195:23) at EmberOneWaitress.startWaitingFor (/app/node_modules/.pnpm/zigbee-herdsman@3.2.0/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:219:50) at /app/node_modules/.pnpm/zigbee-herdsman@3.2.0/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2023:55 at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.0/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20) at EmberAdapter.sendZclFrameToEndpoint (/app/node_modules/.pnpm/zigbee-herdsman@3.2.0/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1963:16) at Request.send (/app/node_modules/.pnpm/zigbee-herdsman@3.2.0/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20) at Endpoint.sendRequest (/app/node_modules/.pnpm/zigbee-herdsman@3.2.0/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:324:20) at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.2.0/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:949:28) at Endpoint.configureReporting (/app/node_modules/.pnpm/zigbee-herdsman@3.2.0/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:686:9) at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.11.0/node_modules/src/lib/modernExtend.ts:158:17))
```

- After the above error, the details screen looks like this (it seems it shows old 'Firmware build date' and 'Firmware version' values; probably because the 'Configuring' step failed?:

[image]

darkxst, 2025-01-01:
@jorisvervuurt Not sure about downgrading bootloader, never tried this. It looks like it flashed ok, but then dongle hasnt installed it. You may need to erase chip first (only possible via SWD/JTAG). However I wouldnt worry about bootloader, having 3.0 shouldn't cause any problems, it should be backwards compatible.

> * After the above error, the details screen looks like this (it seems it shows old 'Firmware build date' and 'Firmware version' values and it reports the dongle as being offline; probably because the 'Configuring' step failed?):

I didnt update firmware build date :( The error might be due to missing the light entity in the converter perhap? I will update converter to test.

Nerivec, 2025-01-01:
Looks like it's failing to setup reporting for availability (missing `zigbee_reporting` component?).
Firmware build date/version will only update with a re-interview (_goes for any device updated through any other mean than Z2M>OTA_). That information is just not retrieved automatically by Z2M, nor advertised by devices.

jorisvervuurt, 2025-01-02:
@Nerivec I deleted the device (not force delete) before re-pairing. That should definitely trigger an interview. ;-)

darkxst, 2025-01-02:
Here is another build, that should fix that error:
https://github.com/darkxst/silabs-firmware-builder/actions/runs/12575715511

jorisvervuurt, 2025-01-02:
Cool, that one indeed fixes the error during the 'Configuring' step and it now correctly shows as 'Online'. 'Exposes' tab only shows LQI but that's as expected if I'm correct? LED behavior is the same as well. 'Firmware build date' and 'Firmware version' still show the old values though (after removing and re-pairing), is that correct?

[image]

jorisvervuurt, 2025-01-02:
By the way, I noticed you (@darkxst) have also started working in a `2012.12.0` branch. In the manifest files, you still use `gpioPortA` and `gpioPortB`. I noticed @Nerivec has replaced those: https://github.com/Nerivec/silabs-firmware-builder/commit/8dc95f4a36158216f9b8351e733c6a462c0b187b#diff-5ebd59bd4d4c122c4bd12393c5243eb5a71562bc73b1b1222c89b606171ecb7d

Just a FYI. ;-)

Nerivec, 2025-01-02:
When you delete a device in Z2M, it is kept in memory until shutdown to allow quick re-add (for example for misbehaving devices that drop then come back a few seconds later). I'd have to check exactly which properties are always re-retrieved, but unless you restart Z2M after the delete, it is not fully deleted, hence not the same behavior as a full interview (the re-interview button forces a full interview).

jorisvervuurt, 2025-01-02:
Aha, good to know!

darkxst, 2025-01-04:
> By the way, I noticed you (@darkxst) have also started working in a `2012.12.0` branch.

Yes, needs a bit more testing but will do a "beta" release soon.

> In the manifest files, you still use `gpioPortA`, `gpioPortB` etcetera. I noticed @Nerivec has replaced those: [Nerivec@8dc95f4#diff-5ebd59bd4d4c122c4bd12393c5243eb5a71562bc73b1b1222c89b606171ecb7d](https://github.com/Nerivec/silabs-firmware-builder/commit/8dc95f4a36158216f9b8351e733c6a462c0b187b#diff-5ebd59bd4d4c122c4bd12393c5243eb5a71562bc73b1b1222c89b606171ecb7d)

This is purely cosmetic, both variants compile down to the same values. I'd prefer to keep a minimal diff, until I decide how to handle multiple branches, for example building v8.0.3 in parallel to v8.1.1 when they released .

darkxst, 2025-01-08:
Closing this as fixed. Latest router builds are now available on both mine and the official SMLIGHT web flasher

closed 2025-01-08 by darkxst