darkxst/silabs-firmware-builder #172: Webflasher fails to connect to ZBDongle-E with router firmware

ditschi, 2025-01-09:
I used the Web Flasher to flash the [router firmware](https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master/Dongle-E/Router) on my ZBDongle-E.  Using 'Custom firmware' that was easy and straight forward.  :+1: 

When connecting the dongle to a USB  Power supply I can also add the router to my zigbee2mqtt which tells me the Device is functional and running as intended. 

When trying to convert back from router to the coordinator firmware the Flasher is failing to connect to the device.  (See  full attached log)

i guess this is the issue 

```
DEBUG Failed to parse frame bytearray(b'\x80\x98\x06\x06x\x06\x06')
Traceback (most recent call last):
  File "/lib/python3.11/site-packages/bellows/ash.py", line 473, in data_received
    frame = parse_frame(data)
            ^^^^^^^^^^^^^^^^^
  File "/lib/python3.11/site-packages/bellows/ash.py", line 353, in parse_frame
    return frame.from_bytes(data)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/lib/python3.11/site-packages/bellows/ash.py", line 222, in from_bytes
    control, data = cls._unwrap(data)
                    ^^^^^^^^^^^^^^^^^
  File "/lib/python3.11/site-packages/bellows/ash.py", line 159, in _unwrap
    raise ParsingError(
bellows.ash.ParsingError: Invalid CRC bytes in frame bytearray(b'\x80\x98\x06\x06x\x06\x06'): expected 9719, got 0606
```

It would be great if the Flasher would support  to convert a custom firmware back to a "known" one

darkxst, 2025-01-09:
Its not possible for the flasher to support this directly with Sonoff builds. You need to either:
- On Linux, use `universal-silabs-flasher` with option `--bootloader-reset sonoff`
- Or disassemble and use physical boot button to trigger bootloader, while running the webflasher.

LinusCDE, 2025-02-03:
> Its not possible for the flasher to support this directly with Sonoff builds. You need to either

Huge thanks! This saved my rear! I had purchased a 2nd dongle and flashed the router fw on it, not realizing that going back to coordinator might not be as easy as going Coordinator -> Router.

Note for future travellers finding them selves here as well:

- `universal-siliabs-flasher` seems to not work as of this time (pip release). It can probe the stick in router mode, but flashing fails because of an exeeded "PROBE_TIMEOUT" (raising it from 2 to 5 didn't help either). Might be fixed in future, so pay attention to the date of this comment.

- Disassembling and using the boot button worked like a charm. Make sure to hold down the boot button while plugging in. The green led should not light up then. After that the device showed up in the web flasher as "Bootloader (recovery)" and flashing worked perfectly well. :+1:

darkxst, 2025-02-03:
yes `--bootloader-reset` for Sonoff dongles has been around for a long time, I just havent as yet integrated this into web flasher, but may eventually.

closed 2025-12-26 by github-actions[bot]