darkxst/silabs-firmware-builder #175: SLZB-06MG24 support?

stickpin, 2025-01-15:
Hi @darkxst,

Are you planning to add SLZB-06MG24 firmware to the repo?
I finally managed to order one and should get it somewhere next week.

Thanks a lot! :)

darkxst, 2025-01-15:
Yes, it will be added, however I dont have one myself for testing yet.

darkxst, 2025-02-01:
I have merged 06Mg24 builds now. They are for now atleast, should be the same as official builds that ship with the device.

stickpin, 2025-02-01:
Hi @darkxst,

Thanks a lot! 
I will finally receive mine on Monday, there was a delay with the shipment.
I plan to migrate from SLZB-07MG24 to SLZB-06MG24 using Ember ZLI.
Let's hope it works. :) 

p.s. Coffee is on its way... ;)

darkxst, 2025-02-01:
Thanks, I will close this now. If you have any issues with the new firmware feel free to open a new issue.

closed 2025-02-01 by darkxst