darkxst/silabs-firmware-builder #176: Current firmware not recognized with web flasher - Sonoff USB Dongle E V2.

HakonJK, 2025-01-18:
Hi

I just bought a new Sonoff Dongle E, with the CP2102 serial chip. I installed the 6.7.0.0 drivers, as recommended in the web-flasher. I have also flashed the device with the [ncp-uart-sw_EZNet6.10.3_V1.0.1.gbl firmware, using SecureCRT. The device is booted with a green light if I plug it into my computer.

I have tried many different firmwares flashed with SecureCRT, with the same result. 

I am a beginner with both HA and these types of procedures, and I'm not a developer or know a lot about this stuff, so I appreciate your patience. I am simply trying to follow guides online, gathering information through YouTube, issue-threads on GitHub, HA forums etc.

I will add the log file from the web flasher. I hope someone can point out any obvious mistakes. 

[silabs_flasher (4).log](https://github.com/user-attachments/files/18465444/silabs_flasher.4.log)

darkxst, 2025-01-18:
It looks like you might be hitting the Windows reset bug even though you have installed the correct drivers. You can try holding the boot button while connecting device to USB and through the web flasher steps. That should allow the web flasher to work.

HakonJK, 2025-01-18:
Thanks for the quick response, your suggestion got me around the issue. I managed to get through the installation process, seemingly correctly, but if I reconnect to confirm the firmware, it shows the image under, which is not exactly similar to what is shown for some of the YouTube videos explaining this process. The device is _not_ showing a green light when booting either. Is this still correct? 

I added the log-file from this seemingly successful installation. I haven't yet set up my HA for ZB2MQTT yet, as I wanted to make sure the dongle had the correct firmware first. 

[silabs_flasher (6).log](https://github.com/user-attachments/files/18466509/silabs_flasher.6.log)

[image]

darkxst, 2025-01-18:
my firmware doesnt include the LED, as you cant see it normally anyway.

Its only going to show bootloader while boot button is pressed!  There is a test webflasher at the below link, that should detect firmware version without boot button (but still needs boot button to flash if reset bug is occuring)
https://github.com/darkxst/cp-test

Best to just try in Z2M, if it starts then you are good and can check the firmware version in Z2M itself.

HakonJK, 2025-01-18:
Great, I'll give it a try, and then proceed with the installation process in Home Assistant. Thanks so much for your help!

darkxst, 2025-02-04:
Closing as complete. Feel free to open a new issue if you have further issues.

closed 2025-02-04 by darkxst