darkxst/silabs-firmware-builder #177: SLZB-07 router build. Pairing

kwanchen, 2025-01-19:
Hi,

I've been using my SLZB-07 as a router to extend my network. Recently, I updated the router firmware (tried both official SMLIGHT as well as yours) and it installed successfully. I thought it would be simple to pair it again to zigbee2mqtt after I moved it to another place so I hit the "FORGET" button on the webUI. However, when I go to the SMLIGHT reconnect router device, I am unable to get it into pairing mode.

Is there a way to force the SLZB-07 to go into pairing mode? I don't recall it being this complicated when I first paired - just went to "Permit Join" and plugged the slzb into the wall. Now I just get a green LED blip when it plugs in, nothing shows up on zigbee2mqtt to suggest that the dongle is even trying to communicate with it.

darkxst, 2025-01-19:
The pairing info is kept even when you reflash. It should however re-connect it without having to re-pair.

You can manually force it to go back into pairing mode, by connecting to the serial port with putty or similar program and send command `network leave`. This is the same as what button on the SMLIGHT web flasher does.

Or if its still connected to the network, you can just delete the device (not force delete though) in Z2M and it will re-enter pairing mode as well.

kwanchen, 2025-01-21:
Thank you for the tip!

I'm not exactly sure how I got into the stick via the serial port - did
something with minicom... but I did somehow enter "network leave" and then
a bunch of text flew up.

Seemed to work!

I definitely did "Force Delete" before which is why it didn't re-enter into
pairing mode. Still new to all this home automation.

Thank you for your help. Sent you 3 coffees!

k.

darkxst, 2025-02-04:
Thanks for the coffees, closing as complete.

closed 2025-02-04 by darkxst

zimbo196, 2025-09-22:
Hello,

I have the same problem with my SLZB MR4 – Radio 1 [EFR32MG26] Zigbee Router. How exactly can I force the ‘network leave’ here? Radio 2 is the Zigbee Coordinator.

Thank you for your help!