darkxst/silabs-firmware-builder #179: Flashing the firmware to Sonoff Zigbee 3.0 USB Dongle Plus V2 hangs at 0%

Kariklebi1, 2025-01-29:
I am trying to flash the firmware on my Sonoff Zigbee 3.0 USB Dongle Plus V2 (without Hardware Flow Control) with the Silabs Firmware Builder (https://darkxst.github.io/silabs-firmware-builder/).
Unfortunately, the flashing process always gets stuck at 0% and does not continue. I have followed all the recommended steps, but the problem persists.

[image]

darkxst, 2025-02-01:
Hi,
  This probably the usual bug with windows drivers. You can find more details about workarounds here
https://github.com/darkxst/silabs-firmware-builder/issues/107#issuecomment-2454076415

pinkfloydFR, 2025-02-26:
-Disassemble device (unscrewing the 2 philips screws on the USB-A side is enough)
-Locate the boot and reset button (close to the USB connector, two little grey dots)
-Hold boot button
-Insert device
-Click connect in the web flasher
-Select the right device
-Wait for the dialog to confirm the bootloader version
-Release boot button

Hansaplast, 2025-03-12:
Thank pinkfloydFR !
This and installing the [SiLabs USB driver](https://www.silabs.com/developer-tools/usb-to-uart-bridge-vcp-drivers?tab=downloads) on my Mac (!) fixed it. 😊

Breezy2G, 2025-03-14:
> -Disassemble device (unscrewing the 2 philips screws on the USB-A side is enough) -Locate the boot and reset button (close to the USB connector, two little grey dots) -Hold boot button -Insert [...]

This was the only thing that worked for me, thank you very much! Hope this helps others out too! 👍 😄

FABIEANGUILLAUME, 2025-03-21:
Hi YOU CAN HELP ME I BOUGHT TWO SONOFF DONGLE ZIGBee 3.0 USB Dongle Plus, EFR32MG21 OR I WILL FLASH ONE IN ZIGBEE AND THE OTHER IN MULTI PAN. I tried on windows 10 and 11 but when installing the firmware it remains stuck at 0.0% and it remains like all the time like that and I'm a beginner 

[image]

darkxst, 2025-03-21:
This probably the usual bug with windows drivers. You can find more details about workarounds here
https://github.com/darkxst/silabs-firmware-builder/issues/107#issuecomment-2454076415

Multi-pan firmware is no longer recommended for use, its best to use pure openthread firmware.

FABIEANGUILLAUME, 2025-03-21:
I wanted it in multi-pan because I want to put the zigbee and the thred to connect it to the home assistant home automation box

Breezy2G, 2025-03-21:
> Hi YOU CAN HELP ME I BOUGHT TWO SONOFF DONGLE ZIGBee 3.0 USB Dongle Plus, EFR32MG21 OR I WILL FLASH ONE IN ZIGBEE AND THE OTHER IN MULTI PAN. I tried on windows 10 and 11 but when installing the firmware it remains stuck at 0.0% and it remains like all the time like that and I'm a beginner
> 
> [image]

I was able to get past this, steps taken were taking the cover off. Flashing original firmware (if needed),Then holding the boot button while plugging into usb for a few seconds and it should force it to install the new firmware (for multi-pan/etc).

darkxst, 2025-03-22:
> I wanted it in multi-pan because I want to put the zigbee and the thred to connect it to the home assistant

You should use 2 dongles, one for Zigbee the other for Openthread. Multi-pan is completely unsupported at this point, there will no further updates or bug fixes.

FABIEANGUILLAUME, 2025-03-22:
no, zigbee for MQTT and LE multi pan for zigbee and thread

FABIEANGUILLAUME, 2025-03-22:
> > Bonjour, pouvez-vous m'aider ? J'ai acheté deux clés USB Sonoff ZIGBee 3.0 Plus, EFR32MG21. Sinon, je vais en flasher une en ZIGBEE et l'autre en Multi-Pan. J'ai essayé sous Windows 10 et 11, mais lors de l'installation du firmware, le problème reste bloqué à 0 % et persiste. Je suis débutant.
> > [image]
> 
> J'ai réussi à contourner ce problème en retirant le capot, en installant le firmware d'origine (si nécessaire), puis en maintenant le bouton de démarrage enfoncé pendant quelques secondes tout en branchant l'appareil sur le port USB. L'installation du nouveau firmware devrait alors être forcée (pour les configurations multi-panoramiques, etc.).

which of the two buttons is the one to start it

FABIEANGUILLAUME, 2025-03-22:
It's good, thank you, I found the right button.

closed 2025-12-26 by github-actions[bot]