darkxst/silabs-firmware-builder #188: [REQUEST] EmberZNet Zigbee NCP for Seeed Studio XIAO MG24

StofLE, 2025-02-12:
Hi,

do you think this tiny stick is supportable? Probably it needs another bootloader and program with suitable pinout.
It is mentioned under z2m adapters (custom) but comes with a blink light example. I did't find any information to get it working.

Ressources are under https://wiki.seeedstudio.com/xiao_mg24_getting_started/#for-seeed-studio-xiao-mg24-sense

Thanks for any hints.

Stefan

darkxst, 2025-02-12:
Yes it will work as long as there is a USB-Serial port passed through the debugger.

You can probably build same as MGM240P but change the chip type:

EFR32MG24B220F1536IM48

and UART pins 

PA08 TX 
PA09 RX

darkxst, 2025-02-19:
I added config for this chip, you can try the builds from here. You would need to flash and test bootloader first. You should get a menu on the serial port if bootloader is working (connect with putty or screen etc).

https://github.com/darkxst/silabs-firmware-builder/actions/runs/13424550181

darkxst, 2025-02-20:
You will need to use openocd to flash bootloader
```
openocd -s scripts/ -f interface/cmsis-dap.cfg -f target/efm32s2_g23.cfg -c "flash bank bank0 efm32s2 0x08000000 0x6000 0 0 0 efm32s2.cpu; init; reset_config srst_nogate; program  ../xiao_mg24_bootloader_2.5.2_BL_PC00.hex; reset; exit"
```

Upstream `openocd` doesnt have support for efm32s2 target so I used the packages provided for unbricking device:
https://wiki.seeedstudio.com/xiao_mg24_getting_started/#unbricking

After that you can just use Universal-silabs-flasher to flash zigbee firmware
```
universal_silabs_flasher.flasher INFO Detected ApplicationType.EZSP, version '8.0.2.0 build 397' (8.0.2.0.397) at 115200 baudrate (bootloader baudrate None)
```

darkxst, 2025-02-20:
https://github.com/darkxst/silabs-firmware-builder/releases/tag/20250220

closed 2025-02-20 by darkxst

StofLE, 2025-03-05:
Wow, thanks. Tried it on my test environment (ZHA). and it worked. :-)