darkxst/silabs-firmware-builder #189: [REQUEST]

EvanRens, 2025-02-15:
Please add NAT64 capability to the Multiprotocol OTBR capability. This would allow various Thread/Matter devices (like TADO) to communicate also with their internet backends for firmware updates and like

darkxst, 2025-02-15:
Multiprotocol is deprecated and no longer supported. NAT64 is implemented at the OTBR level not in firmware.

EvanRens, 2025-02-16:
Does that mean instead of using the Silicon Labs Multiprotocol Add-On I should simply install the OpenThread Border Router add-on ?

darkxst, 2025-02-16:
yes you should use the OTBR addon for Thread, and have separate dongles for Zigbee and Thread.

EvanRens, 2025-02-16:
So two sonoff dongles if I want to use both Thread and Zigbee ?

darkxst, 2025-02-16:
yes that is best, sharing a single dongle for both causes too many issues.

closed 2025-02-20 by darkxst