darkxst/silabs-firmware-builder #192: Support for EFR32MG21 embedded on router board

testuser7, 2025-02-17:
Is it possible to add support for a chip that is embedded on another device's board (Linksys MX4200 router). Communication is via UART.
Is there any special configuration needed to add a new device to the list?

darkxst, 2025-02-20:
Can probably add, you would need to know the exact chip model id and the pins used for UART. Also you would need some way to access the zigbee chip, does it setup a network socket for this etc?

testuser7, 2025-02-20:
The chip designation is `MG21A020JIB01`:
[image]

There are two GPIOs available in the system with the description:
- reset
- recovery

HomeKit is running on the manufacturer's software.
Unfortunately I don't have more information.

darkxst, 2025-02-20:
you would still need to know the uart pins, might be able to guess these if you post the a picture of the surrounding pcb. 

However would still need someway to access the chip for flashing and use. Short of installing Openwrt on the device, this may not be possible with the stock firmware?

testuser7, 2025-02-21:
Access from PC can be done via `ser2net`.

MX4200v1:
[image]
MX4200v2:
[image]

darkxst, 2025-02-21:
There are lots of GPIO connected here, more so than required for UART. 

I suppose the 4 traces coming in from the right side could be UART, but that then means both revisions have different pinouts.

You probably should start by seeing if you can activate bootloader (Assuming they even use gecko bootloader). this should give you a text based menu on the UART console.  You need to hold bootloader pin low, while resetting (also low on reset pin).

from v2, bottom left test pad is probably "reset", one of the other test pads should be bootloader pin (i would guess top left PC00, only as that is the default pin.) Maybe these are the system gpio pins you can see.

That might leave RX/TX on PA00 and PA04 (test pads on right).

If they are not using Gecko bootloader, then you would need SWD access to flash a new bootloader this would be pins PA01 and PA02. In both cases these pins disappear to back (or inner) layer of PCB.

If you can get the bootloader then its likely just a case of trial and error until you find the working combination!

testuser7, 2025-02-21:
Thank you for your help. The only photos I found from the other side of the PCB.
MX4200v1:
[image]

MX4200v2:
[image]

As you can see in both cases there are 8 pins that are probably connected to the chip.

I currently do not have access to this device.
However, I do have access to a device with an K32W041 chip, where I am able to switch to programming mode and read/write memory. But I have not found any example anywhere of how to prepare the appropriate configuration to build firmware.

closed 2025-02-21 by darkxst

darkxst, 2025-02-21:
> In the application that is used for communication in the OEM software, I found something like this:

```
[0mgecko_cmd_gatt_discover_characteristics success.
[0mError - gecko_cmd_gatt_discover_characteristics returned %x.
[0mHandle gecko_evt_le_connection_opened event.
[0mHanlde gecko_evt_le_connection_parameters with size %d.
[0mgecko_evt_gatt_procedure_completed_id 
[0mgecko_evt_gatt_characteristic_value_id 
[0mgecko_evt_hardware_soft_timer_id: %d  
[0mgecko_cmd_gatt_discover_primary_services success.
gatt_gecko_cmd.c
```

Gecko is the SDK, the above is related to Bluetooth. There are two types of bootloaders the standalone bootloader that is used with all Zigbee firmware. Or you can use an application based bootloader where its integrated into the application. We need the standalone version, and if its not there, requires SWD to flash new bootloader.

> But I have not found any example anywhere of how to prepare the appropriate configuration to build firmware.

The easiest way is to fork this repo (SiSDK branch), copy an existing manifest and edit with required config. Push the new manifest to Github and let actions build it. Example here:

https://github.com/darkxst/silabs-firmware-builder/pull/190