darkxst/silabs-firmware-builder #194: Cannot flash my Sonoff ZBDongle-E with web flasher

iu85E, 2025-02-28:
Good morning

I hope this is the right project to report such an issue. I'm trying to flash a newly acquired Sonoff ZBDongle-E. Never used,

I tried flashing it from 2 separate computers, both running Windows 11, with the latest versions of Chrome or Edge. Both fail

when flashing. The dongle is plugged into a USB port on the device, not a cable or a USB hub.

Project

I use the url: https://darkxst.github.io/silabs-firmware-builder/

There I select ZBDongle-E and Select. The browser then asks me to select a device, I select the one list

[image]
[image]

[silabs_flasher.2.log](https://github.com/user-attachments/files/19018679/silabs_flasher.2.log)

darkxst, 2025-02-28:
Windows drivers are buggy, you need to install this driver per the note on webflasher main page:

[image]

Alternatively you can use the "Silabs flasher" add-on for Home Assistant (if your running that!)

closed 2025-02-28 by darkxst

iu85E, 2025-03-01:
sa ne marche pas