darkxst/silabs-firmware-builder #197: DONGLE SONOFF ZigBee

FABIEANGUILLAUME, 2025-03-21:
Hi YOU CAN HELP ME I BOUGHT TWO SONOFF DONGLE ZIGBee 3.0 USB Dongle Plus, EFR32MG21 OR I WILL FLASH ONE IN ZIGBEE AND THE OTHER IN MULTI PAN. I tried on windows 10 and 11 but when installing the firmware it remains stuck at 0.0% and it remains like all the time like that and I'm a beginner 

[image]

eerke2021, 2025-03-25:
I was able to do the zigbee but not the openthread on mine. Same error. Tried manually downloading the firmware. Same error in W11 and Linux Mint (added me to groups as needed). In W11 did you install the old driver as recommended on web flasher page. Pretty sure mine failed till I did.

eerke2021, 2025-03-30:
having read other Threads I tried the boatloader option by taking the device apart and holding the bootbutton down till the screen identifying the bootloader was there. Then it loaded an Open Thread firmware and as well as the progress bars now moving the Logs show a firmware was uploaded ( don't know how to tell which one it was but will attach the logs). 

The old firmware was Open Thread 2.4.4.0 and when I plug it in now it still says  #2.4.4.0

How accurate is it at identifying the current firmware. Latest log says:
2025-03-30 00:52:19 emscripten universal_silabs_flasher.flasher[42] INFO Detected ApplicationType.SPINEL, version 'SL-OPENTHREAD/2.4.4.0_GitHub-7074a43e4' (2.4.4.0) at 460800 baudrate (bootloader baudrate 115200)

full Log of OT update that was succesfull:
 Or maybe not. File is very verbose. Haven't figured out how to load a file here.

eerke2021, 2025-03-30:
[silabs_flasher (1).log](https://github.com/user-attachments/files/19523635/silabs_flasher.1.log)

lets hope this doesn't expand.

eerke2021, 2025-03-30:
yay, Im going to try once more having downloaded the 4.4.4 myself,

eerke2021, 2025-03-30:
RCP onbly goes up to 2.4.5. The 4.4.4 confused me. I am now on 2.4.5. Not sure why the webflasher only goes up to 2.4.4 but I seem to have spent a lot of time going from 2.4.4 to 2.4.5