darkxst/silabs-firmware-builder #198: [REQUEST] EmberZNet 7.5.0.0 Zigbee NCP firmware builds (7.5.0 is available in SDK 4.4.6)

Hedda, 2025-04-02:
EmberZNet 7.5.0.0 Zigbee NCP firmware builds (7.5.0 is available in SDK 4.4.6)

* https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.5.0.0.pdf

Note! EmberZNet 7.5 is a patch release and follows version 7.4.5.0 (so it does not have same EZSP protocol version as EmberZNet 7.4).

- _**Zigbee:**_
  - _Limited support of GCC LTO._
  - _Targeted quality improvements and bug fixes._
- _**Multiprotocol:**_
  - _Targeted quality improvements and bug fixes._

* https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.4.6

* https://github.com/SiliconLabs/gecko_sdk/releases

* https://github.com/SiliconLabs/gecko_sdk/blob/gsdk_4.4/protocol/zigbee/app/util/ezsp/ezsp-protocol.h#L33

* https://github.com/SiliconLabs/gecko_sdk/commit/59d2160fe448b13730a91c0f019a8b4c53c43eb2

paolofaz, 2025-05-28:
Last fw is 6 month older… this project is still alive?

darkxst, 2025-05-31:
Yes still alive, Mainly focused on v8 builds these days.

There are 7.5 builds here, I should do a proper release for these soon
https://github.com/darkxst/silabs-firmware-builder/actions/runs/13599640264

KoLSMS, 2025-06-14:
> Where can I find new v8 builds? I see the latest available here is 8.0.2.0.

same question from me, where can i find 8.x.x.x builds?

thx

darkxst, 2025-06-25:
Now testing 8.2.0 builds, these require Zigbee2Mqtt nightly (dev) builds for the moment. Not sure of status in ZHA yet.

https://github.com/darkxst/silabs-firmware-builder/actions/runs/15867797563

Hedda, 2025-06-25:
> Now testing 8.2.0 builds, these require Zigbee2Mqtt nightly (dev) builds for the moment. Not sure of status in ZHA yet.
> 
> https://github.com/darkxst/silabs-firmware-builder/actions/runs/15867797563

zha uses zigpy’s bellows library and it currently supports up to 8.0.x (i.e. EZSP v16 / EZSP PROTOCOL VERSION 16)

https://github.com/zigpy/bellows/issues/662

darkxst, 2025-06-25:
Right, 8.2.0 is using EZSP v17...

darkxst, 2025-06-28:
@tore1980 You need to explicitly disable hardware flow for zbdongle-e, then it should work.

```
port: /dev/ttyUSB0
adapter: ember
baudrate: 115200
rtscts: false
```

You can grab latest builds from here...
https://github.com/darkxst/silabs-firmware-builder/releases/tag/20250627-Next

darkxst, 2025-06-29:
yes same settings as 8.0.3

> When I update to 8.2.0 (zbdonglee_zigbee_ncp_8.2.0.0_sw_flow_115200.gbl) it dosent start.

Can you post full logs (with debug enabled)?

I have no trouble here running it here...
```
[2025-06-29 10:21:52] info: 	zh:ember: Adapter version info: {"ezsp":17,"revision":"8.2.0 [GA]","build":191,"major":8,"minor":2,"patch":0,"special":0,"type":170}
```

It is possible that something in NVM is interfering, you can try wipe it (will require re-pairing) with this:
https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP/nvm3_initfile.gbl

or perhaps try removing Z2M backups and see if that helps...