darkxst/silabs-firmware-builder #199: cannot connecting the dongle

mharisoy, 2025-04-06:
On the website, after choosing the dongle, the message "Connecting... This can take a few seconds." appears.

Then i got the message :
`Connection failed
The running firmware could not be detected.

Make sure the USB port works and if you are using a USB extension cable, make sure the cable can transfer data. Unplug the Sonoff ZBDongle-E and plug it back in to reset and try again.`

[](https://i.imgur.com/8Uv9kDi.png)

I tried with the actual driver (11.4.0) and the old (6.7.6). The issue is the same.

model : ZBDongle-E
FCC ID : 2APN5ZBD-E
SN : 4461504606

log file :

`2025-04-06 18:54:01 emscripten zigpy.serial[42] INFO Using pyserial-asyncio-fast in place of pyserial-asyncio
2025-04-06 18:54:02 emscripten zigpy.appdb[42] DEBUG SQLite version for <webserial_transport.MockSqlite3 object at 0x1214590>: 3.31.1
2025-04-06 18:54:02 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2025-04-06 18:54:02 emscripten zigpy.serial[42] DEBUG Connection made: <webserial_transport.WebSerialTransport object at 0x1089900>
2025-04-06 18:54:02 emscripten universal_silabs_flasher.gecko_bootloader[42] DEBUG Sending data b'\n'
2025-04-06 18:54:02 emscripten universal_silabs_flasher.gecko_bootloader[42] DEBUG Sending data GeckoBootloaderOption.EBL_INFO
2025-04-06 18:54:02 emscripten universal_silabs_flasher.gecko_bootloader[42] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\x11')
2025-04-06 18:54:03 emscripten universal_silabs_flasher.gecko_bootloader[42] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\x11\x1a')
2025-04-06 18:54:03 emscripten universal_silabs_flasher.gecko_bootloader[42] DEBUG Parsing State.WAITING_FOR_MENU: bytearray(b'\x11\x1a\xc1\x02\x02\x9b{~')
2025-04-06 18:54:04 emscripten zigpy.serial[42] DEBUG Waiting for serial port to close
2025-04-06 18:54:04 emscripten webserial_transport[42] DEBUG Closing serial port
2025-04-06 18:54:04 emscripten webserial_transport[42] DEBUG Closed serial port
2025-04-06 18:54:04 emscripten zigpy.serial[42] DEBUG Connection lost: None
2025-04-06 18:54:04 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.CPC at 460800 baud
2025-04-06 18:54:04 emscripten zigpy.serial[42] DEBUG Connection made: <webserial_transport.WebSerialTransport object at 0x21c3c88>
2025-04-06 18:54:04 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2025-04-06 18:54:04 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2025-04-06 18:54:05 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 1 of 4)
2025-04-06 18:54:05 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to parse buffer bytearray(b'x\x00\x80\xf8\x00x\x80\x00\x80\x00\xf8\xf8\x80\xf8\xf8x\xf8'): ValueError('Invalid flag')
2025-04-06 18:54:05 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2025-04-06 18:54:05 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2025-04-06 18:54:05 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to parse buffer bytearray(b'\x00\xf8\x00x\xf8xx\xf8\x80\x80\xf8'): ValueError('Invalid flag')
2025-04-06 18:54:06 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 2 of 4)
2025-04-06 18:54:06 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2025-04-06 18:54:06 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2025-04-06 18:54:07 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 3 of 4)
2025-04-06 18:54:07 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2025-04-06 18:54:07 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2025-04-06 18:54:08 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 4 of 4)
2025-04-06 18:54:08 emscripten zigpy.serial[42] DEBUG Waiting for serial port to close
2025-04-06 18:54:08 emscripten webserial_transport[42] DEBUG Closing serial port
2025-04-06 18:54:08 emscripten webserial_transport[42] DEBUG Closed serial port
2025-04-06 18:54:08 emscripten zigpy.serial[42] DEBUG Connection lost: None
2025-04-06 18:54:08 emscripten universal_silabs_flasher.flasher[42] INFO Probing ApplicationType.CPC at 115200 baud
2025-04-06 18:54:08 emscripten zigpy.serial[42] DEBUG Connection made: <webserial_transport.WebSerialTransport object at 0x11aaf18>
2025-04-06 18:54:08 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2025-04-06 18:54:08 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2025-04-06 18:54:09 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 1 of 4)
2025-04-06 18:54:09 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending frame CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b'')))
2025-04-06 18:54:09 emscripten universal_silabs_flasher.cpc[42] DEBUG Sending data b'\x14\x00\n\x00\xc4U\xd3\x02\x00\x04\x00\x03\x00\x00\x00\xdb\x12'
2025-04-06 18:54:10 emscripten universal_silabs_flasher.cpc[42] DEBUG Failed to send CPCTransportFrame(endpoint=<EndpointId.SYSTEM: 0>, control=196, payload=UnnumberedFrame(command_id=<UnnumberedFrameCommandId.PROP_VALUE_GET: 2>, command_seq=0, payload=PropertyCommand(property_id=<PropertyId.SECONDARY_CPC_VERSION: 3>, value=b''))), trying again in 0.10s (attempt 2 of 4)
[... cut ...]
2025-04-06 18:54:31 emscripten universal_silabs_flasher.spinel[42] DEBUG Failed to send SpinelFrame(header=SpinelHeader(network_link_id=0, flag=2), command_id=<CommandID.PROP_VALUE_GET: 2>, data=b'\x02'), trying again in 0.10s (attempt 4 of 4)
2025-04-06 18:54:31 emscripten zigpy.serial[42] DEBUG Waiting for serial port to close
2025-04-06 18:54:31 emscripten webserial_transport[42] DEBUG Closing serial port
2025-04-06 18:54:31 emscripten webserial_transport[42] DEBUG Closed serial port
2025-04-06 18:54:31 emscripten zigpy.serial[42] DEBUG Connection lost: None
`

Josef-Reichardt, 2025-04-09:
I had the same issue. After a few retries (with open browser terminal) I got the next screen where I was able to choose the new firmware but then it was stuck at 0%. I'm using Google Chrome 135.0.7049.85.

N1CK145, 2025-04-30:
I'm experiencing the same issue on the same device with the same log output. I've tried reconnecting about five times, but it's still not working.

boracakar, 2025-06-29:
I also got the same issue. Sonoff fw update website on the other hand can detect the dongle and also updates it without any issue.