darkxst/silabs-firmware-builder #202: Enable external antenna on XIAO MG24

Imaginous, 2025-05-21:
It would be great if ther would be a version or command to enable the external antenna on the XIAO MG24.

According to the documentation this can be done via:

Method for switching antennas
The Seeed Studio XIAO MG24 has two antenna options: an internal antenna and an external antenna. For convenience, you can choose to use the internal antenna, and to enhance signal strength, you can opt for the external antenna. Below are the method for switching between the two antennas.

PB04 is used to select between using the built-in antenna or an external antenna. Before that, you need to set PB05 high level to turn on this function. If PB04 is set low level, it uses the built-in antenna; if it set to high level, it uses the external antenna. Default is low level. If you want to set it high, you can refer the code below:

#define RF_SW_PW_PIN PB5
#define RF_SW_PIN PB4

void setup() {
  // turn on this antenna function
  pinMode(RF_SW_PW_PIN, OUTPUT);  
  digitalWrite(RF_SW_PW_PIN, HIGH);

  delay(100);

  // HIGH -> Use external antenna / LOW -> Use built-in antenna
  pinMode(RF_SW_PIN, OUTPUT);  
  digitalWrite(RF_SW_PIN, HIGH);

Imaginous, 2025-06-10:
@darkxst 

I managed to build my own firmware and made some changes to /src/zigbee_ncp/app.c. But I don't know how I can only include this code for the UFL version of the Seeedstudio Xiao MG24.
So it can be merged with the main branch.

I also created a nsnifest for it. 

I have included the files I have changed to enable the external U.FL antenna connector and enable hardware RTS/CTS. 

Files are zipped, otherwise I couldn't attach them. 

[changes.zip](https://github.com/user-attachments/files/20665939/changes.zip)

darkxst, 2025-06-26:
Given all builds share the same project sources, this would need to be applied as a patch file during the build process.

Probably need to add a "patch" field to the manifest, then patch `build_project.py` to read and apply this patch during setup for build.

Btw, there is no evidence that building Zigbee NCP at 230400 helps performance, and in fact it has caused some weird bugs along the way.

Imaginous, 2025-06-26:
Is there a "how to" to make a patch?
Or are you willing to make a patch? 

I haven't run into issues (yet) with the higher baud rate, but don't mind if it's lowered to 115200. 

This patch system is new to me, I'm an old (school) developer ;)