darkxst/silabs-firmware-builder #213: Accidentally flashed the wrong firmware, can't seem to reflash with boot mode

rasika-jay, 2025-07-10:
I accidentally flashed my `slzb07Mg24` (07) dongle with `slzb06Mg24` firmware and when I use the bootloader moder (`flash`+`gnd`) mode, it looks like it's flashing and goes to 100%, but it doesn't work after that.

Anyone know how to recover? is it completely dead?

rasika-jay, 2025-07-11:
hey @tl-sl , thank you so much! - that worked, i'm back up and running.

although I flashed `EFR32MG24A020F1024IM40_app_clear_134217728_1048576_8192_134242304.gbl` as per the doc: https://github.com/Nerivec/silabs-firmware-recovery/tree/main

mattisz, 2025-07-15:
Sorry, I meant to @rasika-jay

rasika-jay, 2025-07-15:
> Sorry, I meant to [@rasika-jay](https://github.com/rasika-jay)

what dongle are you using?

First, I put mine in to bootloader mode by shorting the pins (See: https://smlight.tech/manual-slzb-07/) using tweezers, I then remove the tweezer once plugged in, but before the flash process.

I then erased using:
`universal-silabs-flasher --device /dev/<port> flash --firmware EFR32MG24A020F1024IM40_app_clear_134217728_1048576_8192_134242304.gbl`

once that finished (without unplugging), I then flashed the intended firmware:
`universal-silabs-flasher --device /dev/<port> flash --firmware slzb07Mg24_zigbee_ncp_8.0.3.0_115200.gbl`

tl-sl, 2025-07-16:
> /universal-silabs-flasher --bootloader-reset rts_dtr 

For Sonoff SLZB-07* devices you need (can be used instead of the tweezer hack)

```
/universal-silabs-flasher --bootloader-reset slzb07 <rest of args>
```

mattisz, 2025-07-16:
Turns out the SLZB-07MG24 I bought was actually an SLZB-07. Confirmed by looking at the chip. Unfortunately, I was unable to recover it after flashing the app_clear for SLZB-07MG24.

tl-sl, 2025-07-16:
@mattisz if it is SLZB-07, you need to flash app_clear for EFR32MG21!

closed 2025-12-28 by github-actions[bot]