darkxst/silabs-firmware-builder #216: Flashing SLZB-07 claims to succeed, but comes back with old version number

Ablu, 2025-08-05:
I am trying to swtich my SLZB-07 to router firmware. The flashing process seemingly works OK, but the device does not actually seem to start running the router firmware. It comes back always with the same firmware regardless of what I attempt to flash:

Zigbee (EZSP) 7.3.0.0 build 131
SMLIGHT SLZB-07

I also tried to forcefully enter the flashing mode. That then reports from bootloader prior to flashing. Flashing from that state has the same effect: the new firmware seemingly never takes effect.

Flashing log: 

[silabs_flasher (2).log](https://github.com/user-attachments/files/21597798/silabs_flasher.2.log)

Ablu, 2025-08-05:
> How old is your device?

I bought it a few months ago and had it gathering dust until I picked it up. I tried updating with https://smlight.tech/flasher/ too and it has the same behavior. I also tried the "unlock bootloader" option on your version. That gave me a "Failed" but then reported a version from bootloader on next flash (which again did nothing)

Board is labeled with `[07] r1.91`

Is there any way I can confirm the bootloader lock status?

darkxst, 2025-08-05:
> Board is labeled with `[07] r1.91`

OK, This wont be affected by that old bootloader then. 

However I not too sure why it would then fail, when apparently working. I didnt see any errors in your log. Have tried to flash a different build, like Openthread?

closed 2025-12-23 by github-actions[bot]