darkxst/silabs-firmware-builder #222: Aeotec Zi-Stick ZGA008 bricked after firmware flash, cannot recover (no reset button, XMODEM loop)

andruwa13, 2025-09-18:
Device: Aeotec Zi-Stick (ZGA008)

Problem summary:
- Bricked after attempted firmware flash (tried both Zigbee and Thread .gbl, including official ZGA008 Zigbee 7.4.4)
- The stick shows up as /dev/ttyUSB0, but can't be probed by Elelabs_EzspFwUtility.py or universal-silabs-flasher.
- Attempting any flash/utility probe: "Couldn't communicate with the adapter in Zigbee (EZSP) mode, Thread (Spinel) mode or bootloader mode"
- Every flash attempt triggers: "Successfully restarted into X-MODEM mode! Starting upload of the new firmware... DO NOT INTERRUPT(!)", but after a while: "Failed to restart into bootloader mode"
- No serial output in screen/minicom at 115200/230400, even after disconnect/reconnect or trying alternative ports.
- The stick has NO reset button or obvious pads, only G R C D 3V debug points visible on PCB (see attached photo). 
- Physically shorting R-G doesn't reset or revive the stick.
- Multiple systems tested (Raspberry Pi, Linux desktop), multiple cables/ports.

Steps already done:
- Tried all available and official firmware for ZGA008 (Zigbee, Thread, NCP, different baudrates)
- Flashed with both Elelabs_EzspFwUtility.py and universal-silabs-flasher (latest versions)
- Serial console stays silent in XMODEM mode, [screen is terminating] if device not recognized
- Verified no processes occupying /dev/ttyUSB0
- Every flash command enters XMODEM, seemingly uploads, then fails with bootloader mode restart
- Physically shorted R-G for hard reset: no effect
- Device is seen by dmesg as USB-UART bridge, but probe/flash never succeeds

Request:
- Is there any way to force recovery for this particular Aeotec ZGA008 revision?
- Are there working steps for deep unbrick (JTAG/SWD or other ISP via debug pads)?
- Can you provide updated recovery instructions or a specific .gbl or tool to escape XMODEM/bootloader loop for this stick variant?

Attachments: Adding a PCB photo (see below for reference).

Thank you!

andruwa13, 2025-09-18:
Additional update:

Just tried again with this command (on macOS):
universal-silabs-flasher --device /dev/tty.usbserial-1420 --probe-method bootloader flash --firmware /Users/ex/Downloads/ELU0141_MG21_ot_rcp_123_211204.gbl
Result:
2025-09-18 10:51:17.169 localhost universal_silabs_flasher.flash INFO Failed to read firmware metadata: {exc!r}
2025-09-18 10:51:17.170 localhost universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud

Also, apologies — my initial log/photo attachments did not go through on my first comment. I have now re-uploaded these files in previous comments for your reference. If there are still issues with attachments, please let me know and I can provide them directly.

andruwa13, 2025-09-18:
Thank you for your reply and for clarifying that the device can likely always be recovered via bootloader. However, could you please specify the correct procedure for forcing this specific Aeotec Zi-Stick ZGA008 into bootloader mode? There is no physical reset button or BOOT pin accessible on this revision, and shorting the visible R-G pads on the PCB hasn't worked. 

Any detailed steps or tricks for entering bootloader with this hardware (especially after sequential firmware flash failures) would be greatly appreciated. 

With respect.

darkxst, 2025-09-18:
I have not used this device myself, however my guess would be: Short the R-G pads before plugging the device into USB port, then check serial console for menu.

I would also guess the other pads (C)lock and (D)ata are for SWD which can also be used for flashing but requires suitable hardware debugger.

andruwa13, 2025-09-18:
Unfortunately, nothing happens when I short R and G while plugging in the stick – there is never any bootloader output or text in minicom/screen at any baud rate, and the device status in minicom just stays 'Offline'.

When I attach the device with R and G shorted, the system log shows:

```
usb 1-1.3: new full-speed USB device number 17 using xhci_hcd
[64476.241618] usb 1-1.3: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[64476.241656] usb 1-1.3: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[64476.241671] usb 1-1.3: Product: USB Serial
[64476.246455] ch341 1-1.3:1.0: ch341-uart converter detected
[64476.252834] usb 1-1.3: ch341-uart converter now attached to ttyUSB0
```

So the system detects a CH341 USB-UART bridge and assigns it to ttyUSB0, but there is never any response from the stick, no textual output, and no sign of bootloader mode no matter what I try. If you have any further ideas, I would really appreciate them. Thank you!

andruwa13, 2025-09-18:
1. I checked the stick operation; it was detected as ttyUSB0. Screenshot '1.jpg' attached.
2. I updated the firmware with the command:
```
python3 Elelabs_EzspFwUtility.py ele_update -p /dev/ttyUSB0 -v zigbee
```
3. Then I used the web flasher (https://darkxst.github.io/silabs-firmware-builder/) and upgraded to version 7.4.4.0 (selected ZB-GW04 v1.2, EZSP 6.10.3.0 for EFR32MG21 — see screenshot '2.jpg').

After this step, the device stopped working correctly: it cannot be connected, reflashed, or detected (no serial output, neither Elelabs nor universal-silabs-flasher can detect it, stuck in XMODEM loop).

Attaching both screenshots for clarity.

[image]

andruwa13, 2025-09-22:
Thank you very much for the support and all proposed recovery options! I will keep experimenting with different recovery methods and update the thread if anything works. I have also contacted Aeotec technical support about 10 minutes ago—once I get any news or answers, I'll share an update here.

gonzague, 2025-12-11:
Hello

Running into the same problem as you, or at least similar: I flashed the ZGA008 and then .. I cant use it :( 

```
universal-silabs-flasher --device /dev/tty.usbserial-130 flash --firmware ncp-uart-hw-v7.4.5.0-aeotec-zga008-115200.gbl
2025-12-11 18:16:51.115 Mac universal_silabs_flasher.flash INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=1, sdk_version='4.4.5', ezsp_version='7.4.5.0', ot_rcp_version=None, cpc_version=None, fw_type=<FirmwareImageType.ZIGBEE_NCP: 'zigbee_ncp'>, fw_variant=None, baudrate=115200)
2025-12-11 18:16:51.115 Mac universal_silabs_flasher.flasher INFO Probing ApplicationType.EZSP at 115200 baud
2025-12-11 18:16:52.784 Mac universal_silabs_flasher.flasher INFO Detected ApplicationType.EZSP, version '6.10.3.0 build 297' (6.10.3.0.297) at 115200 baudrate (bootloader baudrate None)
ncp-uart-hw-v7.4.5.0-aeotec-zga008-115200.gbl  [####################################]  100%   
```

Probing afterwards.. well.. returns nothing  : 
```
universal-silabs-flasher --device /dev/tty.usbserial-130 probe
2025-12-11 18:30:24.088 Mac universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2025-12-11 18:30:26.597 Mac universal_silabs_flasher.flasher INFO Probing ApplicationType.EZSP at 115200 baud
2025-12-11 18:30:34.613 Mac universal_silabs_flasher.flasher INFO Probing ApplicationType.EZSP at 460800 baud
2025-12-11 18:30:42.630 Mac universal_silabs_flasher.flasher INFO Probing ApplicationType.SPINEL at 460800 baud
2025-12-11 18:30:51.356 Mac universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 460800 baud
2025-12-11 18:30:56.182 Mac universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 115200 baud
2025-12-11 18:31:01.006 Mac universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 230400 baud
2025-12-11 18:31:05.827 Mac universal_silabs_flasher.flasher INFO Probing ApplicationType.ROUTER at 115200 baud
Error: Failed to probe running application type
```

gonzague, 2025-12-14:
Hey @Nerivec thanks a lot, I tried the boot loader reset methods with no luck .. I ended up sending the stick back to Amazon